platformio/platformio-core #1337: Can't build project now (2 computers no changes)

TPLindley, 2018-02-06:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

TPLindley, 2018-02-06:
I am using Visual Studio Code v1.19.3 and PlatformIO IDE Home: 0.6.1 Core: 3.5.2a7 on two computers. I went to compile my ESP8266 application today and both computers will not build the solution. I am getting a strange error (below). I tried uninstalling PIO and re-installing but that didn't make a difference.

I am trying to program a new batch of chips so really need to get this moving. Thanks.

Error: Traceback (most recent call last):
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 80, in cli
    check_project_envs(config, environment or env_default)
  File "c:\users\tom_1prp9sf\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 400, in check_project_envs
    unknown = set(environments) - known
TypeError: 'NoneType' object is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

AMoo-Miki, 2018-02-06:
FYI this works just fine:
```
pip install -U git+https://github.com/platformio/platformio-core.git@e5543b2aee5359dff29b432b38c9ade0e82ae91e#egg=platformio-core --process-dependency-links
```

which means 1362630 is what caused the breakage.

ivankravets, 2018-02-06:
Sorry! Fixed in the latest build. Please open PlatformIO IDE Terminal and type `pio upgrade --dev`

closed 2018-02-06 by ivankravets

TPLindley, 2018-02-08:
Thanks! Worked like a champ.