platformio/platformio-core #1338: Cannot build for any platfrom

benchuk, 2018-02-06:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.installed all plugins
2.updated everything
3.get an error

### Actual Results
Error: Traceback (most recent call last):
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 80, in cli
    check_project_envs(config, environment or env_default)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 400, in check_project_envs
    unknown = set(environments) - known
TypeError: 'NoneType' object is not iterable

Severity | Provider | Description | Line
-- | -- | -- | --

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

viktak, 2018-02-06:
Same error here after updating platformio-ide to the latest this morning.

thehaxxa, 2018-02-06:
Same issue for both VScode & Atom

janosadelsberger, 2018-02-06:
Yep, doesn't work here either.

ivankravets, 2018-02-06:
Sorry! Fixed in the latest build. Please open PlatformIO IDE Terminal and type `pio upgrade --dev`

closed 2018-02-06 by ivankravets

benchuk, 2018-02-07:
Still happens to me...

benchuk, 2018-02-07:
Error: Traceback (most recent call last):
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 80, in cli
    check_project_envs(config, environment or env_default)
  File "/Users/ben/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 400, in check_project_envs
    unknown = set(environments) - known
TypeError: 'NoneType' object is not iterable

benchuk, 2018-02-07:
Bens-MacBook-Pro:180206-090214-nanoatmega328 ben$ pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 3.5.2a8
Release notes: http://docs.platformio.org/en/latest/history.html

ivankravets, 2018-02-07:
Try again. Should work with 3.5.2a8

mishagin-tim, 2018-02-08:
Have the same issue after upgrade.

`MacBook-Air-Timur:180208-163410-uno timurmisagin$ sudo pio upgrade --dev
Password:
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 3.5.2a8
Release notes: http://docs.platformio.org/en/latest/history.html`

ivankravets, 2018-02-08:
@mishagin-tim could you provide a full output with error?

mishagin-tim, 2018-02-08:
Sure 
'''
*******************************************************************************************************************************************************************
Obsolete PIO Core v3.5.2a7 is used (previous was 3.5.2a8)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*******************************************************************************************************************************************************************
Error: Traceback (most recent call last):
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 80, in cli
    check_project_envs(config, environment or env_default)
  File "/Users/timurmisagin/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 400, in check_project_envs
    unknown = set(environments) - known
TypeError: 'NoneType' object is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
'''

mishagin-tim, 2018-02-08:
I've not seen the yellow caption before.
Gonna try to remove core duplicate from my computer. See what happens.

mishagin-tim, 2018-02-08:
Yep, it works. Thank you for feedback!