platformio/platformio-core #1339: Home: Could not list directory /home/username

artbody, 2018-02-06:
Could not list directory by  /home/username
Import a Arduino Projekt
Choose a directory with existing Arduino IDE project:
doesn't work
BUT
home/username/Documents/PlatformIO/Projects
works
So why doesn't home/username not list directory ???

WORKAROUND was to copy my files to home/username/Documents/PlatformIO/Projects
but that's not professional - > and arduino saves the projects by default in /home/username/Arduino
but normally you could save  files, projects or what ever, everywhere you want to

ivankravets, 2018-02-06:
It SHOULD work. Which errors do you see?

artbody, 2018-02-06:
`Could not list directory by /home/username`

ivankravets, 2018-02-13:
Maybe, permission issue?

Could you try in Terminal?
```
python -c "import os; print os.listdir(os.path.expanduser('~'))"
python -c "import os; print os.listdir('/home/username')"
```

artbody, 2018-02-15:
```
 python -c "import os; print os.listdir(os.path.expanduser('~'))"
  File "<string>", line 1
    import os; print os.listdir(os.path.expanduser('~'))
                      ^
SyntaxError: invalid syntax

```
 ```
python -c "import os; print os.listdir('/home/username')"
  File "<string>", line 1
    import os; print os.listdir('/home/username')
                      ^
SyntaxError: invalid syntax
```
ok
python console
```
 python
Python 3.5.4 (default, Feb  6 2018, 02:10:34) 
[GCC 6.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> print os.listdir('/home/username')
  File "<string>", line 1
    import os; print os.listdir('/home/username')
                      ^
SyntaxError: invalid syntax
>>>  os.listdir('/home/username')
['some.sh', '.mplayer', ..... '.hydrogen']
```
username was replaced by my username 

so i think python and the import os module works

ivankravets, 2018-02-15:
Oh... Python 3. See updated commands https://github.com/platformio/platformio-core/issues/1339#issuecomment-365440763

ivankravets, 2018-02-19:
@apenaz artbody please provide an output of
```
ls -al /home
```

closed 2018-03-06 by ivankravets