platformio/platformio-core #1343: platformio core run create a unwanted buffer on vim

onekk, 2018-02-07:
### Configuration

**Linux Debian Stretch**:

**PlatformIO, version 3.5.1**

**VIM 8.0**

### Description of problem
Platformio run with the command:

`platformio -f -c vim run`

create an unwanted buffer after the run with filename 
`[Wed Feb 7 19`

That is the first part of the line

`[Wed Feb  7 19:08:57 2018] Processing `

The quickfix system of VIM maybe interpreter the 19:08:57 as a part of an error line.

sillyfrog, 2018-06-15:
As per my post here: https://community.platformio.org/t/running-make-in-vim-replaces-the-open-file/4363/2 I'm having the same issue.

Having a quick look at the code, applying this patch sorted it:
```
--- /tmp/run.py	2018-06-15 13:32:12.000000000 +1000
+++ run.py	2018-06-15 13:35:21.000000000 +1000
@@ -198,7 +198,7 @@

         if not self.silent:
             click.echo("[%s] Processing %s (%s)" %
-                       (datetime.now().strftime("%c"),
+                       (datetime.now().strftime("%c").replace(":", "-"),
                         click.style(self.name, fg="cyan", bold=True),
                         "; ".join(env_dump)))
             click.secho("-" * terminal_width, bold=True)
```

I had a bit of a look to see if I could extract the `caller` attribute, but only had a few minutes and couldn't figure out the best (any) way to get it with out a lot of code changes (in my quick look). If it's possible to get the `caller` from the `cmd_ctx` or similar, I'm happy to do up a PR for this.

Cheers.

ivankravets, 2018-06-15:
Please re-test with `pio upgrade --dev`

commit 389783ad (2018-06-15): Removed "date&time" when processing project with "platformio run" command // Resolve #1343
commit 6f905e31 (2018-06-15): Merge branch 'hotfix/v3.5.4' into develop

* hotfix/v3.5.4:
  Removed "date&time" when processing project with "platformio run" command // Resolve #1343
  Improve documentation for advanced scripting
closed 2018-06-15 by ivankravets