platformio/platformio-core #1344: LDF not working correct

serioustk, 2018-02-08:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.2a8

### Description of problem
I experience problems with the LDF. I hope, this issue is in the right place here.

The desired library ```config.h``` is shown in the dependency graph but using its elements leads to a compilation error (see below)

I am using this folder structure (pretty standard):
```
├── lib
│   ├── config
│   │   └── config.h
│   └── readme.txt
├── platformio.ini
└── src
    └── main.cpp
```

#### Steps to Reproduce

1. Create files
2. Paste content
3. Build

### Actual Results
```
[02/08/18 08:06:06] Processing nodemcuv2 (build_flags: -std=c++11; lib_deps: Blynk, PCA9685; platform: espressif8266; board: nodemcuv2; framework: arduino; lib_ldf_mode: chain+)
----------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Platform: Espressif 8266 :: NodeMCU 1.0 (ESP-12E Module)
System: ESP8266 80MHz RAM/80KB Flash/4MB
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain+ Compatibility/light
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> v0.5.0
|-- <PCA9685>
|-- <config>
Compiling .pioenvs\nodemcuv2\src\main.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkDebug.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkHandlers.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkTimer.o
src\main.cpp: In function 'void loop()':
src\main.cpp:9:20: error: 'MY_UNIQUE_CONST' was not declared in this scope
Serial.println(MY_UNIQUE_CONST);
^
*** [.pioenvs\nodemcuv2\src\main.o] Error 1
======================== [ERROR] Took 3.79 seconds ========================
```

### Expected Results
Successful build

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -std=c++11
lib_ldf_mode = chain+
lib_deps =
    Blynk
    PCA9685
```

**Source file to reproduce issue:**
****```main.cpp```****
```cpp
#include <Arduino.h>
#include <config.h>

void setup() {
    Serial.begin(115200);
}

void loop() {
    Serial.println(MY_UNIQUE_CONST);
}
```

****```config.h```****
```cpp
#ifndef config_h
#define config_h

#define MY_UNIQUE_CONST 5

#endif
```
### Additional info

ivankravets, 2018-02-08:
```
lib_deps =
    config
    Blynk
    PCA9685
```

Does it work?

serioustk, 2018-02-08:
No, it doesn't. The error still persists:

```
[02/08/18 14:55:02] Processing nodemcuv2 (build_flags: -std=c++11; lib_deps: config, Blynk, PCA9685; platform: espressif8266; board: nodemcuv2; framework: arduino; lib_ldf_mode: chain+)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'config'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'config'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
Platform: Espressif 8266 :: NodeMCU 1.0 (ESP-12E Module)
System: ESP8266 80MHz RAM/80KB Flash/4MB
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain+ Compatibility/light
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> v0.5.0
|-- <PCA9685>
|-- <config>
Compiling .pioenvs\nodemcuv2\src\main.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkDebug.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkHandlers.o
Compiling .pioenvs\nodemcuv2\lib3d7\Blynk_ID415\utility\BlynkTimer.o
src\main.cpp: In function 'void loop()':
src\main.cpp:9:20: error: 'MY_UNIQUE_CONST' was not declared in this scope
Serial.println(MY_UNIQUE_CONST);
^
*** [.pioenvs\nodemcuv2\src\main.o] Error 1
========================================================================================== [ERROR] Took 5.73 seconds ==========================================================================================
```

platform.ini:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -std=c++11
lib_ldf_mode = chain+
lib_deps =
    config
    Blynk
    PCA9685
```

ivankravets, 2018-02-08:
Blynk library does not depend on "config" library. I recommend doing configuration in `platformio.ini`

```ini

[common]
config = 
  -D MY_UNIQUE_CONST=5

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags =
  ${common.config}
  -std=c++11
lib_ldf_mode = chain+
lib_deps =
    config
    Blynk
    PCA9685
```

serioustk, 2018-02-08:
I think we are missunderstanding each other: the problem persists in LDF. I have broken it down to the smallest example I can imagine:

****```main.cpp```****
```cpp
#include <config.h>

void setup() {
}

void loop() {
    int a = MY_UNIQUE_CONST;
}
```

****```config.h```****
```cpp
#ifndef config_h
#define config_h

#define MY_UNIQUE_CONST 5

#endif
```

****```platform.ini```****
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
```

According to ```readme.txt``` file in ```lib``` folder this should work, but in fact, it fails with the following error:
```
[02/08/18 15:21:27] Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Platform: Espressif 8266 :: NodeMCU 1.0 (ESP-12E Module)
System: ESP8266 80MHz RAM/80KB Flash/4MB
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain Compatibility/light
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <config>
Compiling .pioenvs\nodemcuv2\src\main.o
Archiving .pioenvs\nodemcuv2\libc52\libconfig.a
Archiving .pioenvs\nodemcuv2\libFrameworkArduinoVariant.a
Compiling .pioenvs\nodemcuv2\FrameworkArduino\Esp.o
Compiling .pioenvs\nodemcuv2\FrameworkArduino\FS.o
Compiling .pioenvs\nodemcuv2\FrameworkArduino\FunctionalInterrupt.o
src\main.cpp: In function 'void loop()':
src\main.cpp:7:13: error: 'MY_UNIQUE_CONST' was not declared in this scope
int a = MY_UNIQUE_CONST;
^
*** [.pioenvs\nodemcuv2\src\main.o] Error 1
========================================================================================== [ERROR] Took 1.97 seconds ==========================================================================================
```

vanklompf, 2018-02-27:
I have similar problem. It seems that opposite to what [LDF manual](http://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf) is saying, lib directory within project is not scanned at all.

vanklompf, 2018-02-27:
Ok, I have investigated this issue:

1. For libraries in LibraryManager dir ArduinoLibBuilder is used, but for libraries in project ```lib``` dir PlatformIOLibBuilder is used. I'm not sure if it makes any difference.
2. If library has ```src``` subdir (even if empty), than it is scanned instead of root directory of library. **This was actual root cause of my issue.**

I'm not sure if this is actual bug or not. Looking on some examples it seems that some libraries have code in ```src``` and other in root. But probably never in both. I guess that some parameter in ```library.json``` that shows location of "public" headers in library might be useful. Arduino ```library.properties``` has similar concept.

ivankravets, 2018-03-02:
If you use `library.proeprties` or have Arduino-based library, ArduinoLibBuilder will be used by default. See specification https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files

You can use any library structure with PlatformIO. Just add `library.json` manifest to the root of library.

vanklompf, 2018-03-02:
```library.json``` was there. It was [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) library. It seems that Arduino ```library.properties``` have precedence over PlatformIO ```library.json```?

ivankravets, 2018-03-06:
`library.json` should be used firstly. See https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piolib.py#L41

Neeco1, 2018-04-05:
Hello,

I did not want to open a new issue for this, but I am experiencing similar problems with the LDF regarding defined preprocessor constants.

My `platformio.ini` looks like this:
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_ldf_mode = deep+
```

This `main.cpp` compiles just fine as expected:
```c
#include <Arduino.h>

#define BOARD_ESP32     1
#define BOARD_ESP8266   2

#define MY_PLATFORM 1

#if MY_PLATFORM == BOARD_ESP32

    #include <Basecamp.hpp>

#endif

void setup() { }
void loop() { }
```

while this does not compile: 
```c
#include <Arduino.h>

#define BOARD_ESP32     1
#define BOARD_ESP8266   2

#define MY_PLATFORM BOARD_ESP32

#if MY_PLATFORM == BOARD_ESP32

    #include <Basecamp.hpp>

#endif

void setup() { }

void loop() { }
```
I receive the error `src/main.cpp:10:28: fatal error: Basecamp.hpp: No such file or directory`.
This is kind of strange, since the Basecamp library is still in the same place. The only difference now is that I re-used the preprocessor constant `BOARD_ESP32` as a value for `MY_PLATFORM`. This should not affect the include of the Basecamp library.

Any ideas on this? Thanks in advance!

TD-er, 2018-07-05:
Is it possible in the last few days, something has changed in the PlatformIO code handling the platformio.ini file?

It looks like the set build_flags are not processed properly, or at least somewhere cached?
I have a rather elaborate setup with ESP32 and ESP8266 configurations in the same .ini file and when I try to build a configuration for ESP8266, it also tries to compile code for ESP32. Even though this code is surrounded by `#ifdef ESP32` and `#endif`.

The color coding in Visual Studio Code also suggest those flags are active.
I am running PlatformIO "home 0.9.6", "Core 3.6.0a" and 
ESP32:  `platform                  = espressif32@1.1.1`
ESP82xx: `platform                  = espressif8266@1.7.3`

This is running on Windows, but [Travis](https://travis-ci.org/letscontrolit/ESPEasy/builds/400226657?utm_source=github_status&utm_medium=notification) had the same issues last night and that's running on some Linux environment I guess.  (ESP32 run in Travis output is not related to this issue, but the other build errors are)

ivankravets, 2018-07-08:
You use default LDF mode which does not handle C preprocessor's conditions. See https://github.com/letscontrolit/ESPEasy/blob/mega/platformio.ini#L89

Please use `lib_ldf_mode = chain+`. See http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

ademyankov, 2019-05-09:
This LDF dude doesn't even work with the libraries from platformio library registry!!!

It looks like this LDF issue is a quite persistent one and really annoying. Primitive project setup is already taking 1 day from me and it still doesn't work!

I have the same issue with FreeRTOS library. No matter what I tried: lib_ldf_mode, rebuild indexes, specify lib_deps as name or id, whatever... nothing helps!

**platformio.ini**

```
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = stm32cube
board_build.mcu = stm32l475vgt6
board_build.f_cpu = 80000000L
upload_protocol = mbed
monitor_speed = 115200
lib_deps = STM32duino FreeRTOS
lib_ldf_mode = chain+
```

**The library is obviously there:**

[image]

**But I cannot build the code. It doesn't see:**

`#include <STM32FreeRTOS.h>`

**The main is an example from the FreeRTOS library itself:**
```
/*
 * Based on AnalogRead_DigitalRead example from: https://github.com/feilipu/Arduino_FreeRTOS_Library
 * Modified by: Frederic Pillon <frederic.pillon (at) st.com>
 */
#include <STM32FreeRTOS.h>

// If no default pin for user button (USER_BTN) is defined, define it
#ifndef USER_BTN
#define USER_BTN 2
#endif

// Declare a mutex Semaphore Handle which we will use to manage the Serial Port.
// It will be used to ensure only only one Task is accessing this resource at any time.
SemaphoreHandle_t xSerialSemaphore;
```

**Build console:**

```
> Executing task: C:\Users\ademyankov\.platformio\penv\Scripts\platformio.exe run <

Processing disco_l475vg_iot01a (framework: stm32cube; platform: ststm32; board: disco_l475vg_iot01a)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_l475vg_iot01a.html
PLATFORM: ST STM32 > ST DISCO-L475VG-IOT01A
HARDWARE: STM32L475VGT6 80MHz 128KB RAM (1MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain+) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\disco_l475vg_iot01a\src\main.o
src\main.c:5:10: fatal error: STM32FreeRTOS.h: No such file or directory

***********************************************************************
* Looking for STM32FreeRTOS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:STM32FreeRTOS.h"
* Web  > https://platformio.org/lib/search?query=header:STM32FreeRTOS.h
*
***********************************************************************

#include <STM32FreeRTOS.h>
^~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs\disco_l475vg_iot01a\FrameworkBSP\stm32l475e_iot01_magneto.o
Compiling .pioenvs\disco_l475vg_iot01a\FrameworkBSP\stm32l475e_iot01_psensor.o
*** [.pioenvs\disco_l475vg_iot01a\src\main.o] Error 1
Compiling .pioenvs\disco_l475vg_iot01a\FrameworkBSP\stm32l475e_iot01_qspi.o
============================================================================ [ERROR] Took 6.35 seconds ============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

ademyankov, 2019-05-09:
Sorry, I didn't add the version of the software, here are those:
PlatformIO, version 3.6.7
VSCode, version 1.33.1
Windows 10 Enterprise, version 1709, OS Build: 16299.1087

ivankravets, 2019-05-09:
> COMPATIBILITY(soft)
See http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-compat-mode

You are going to use Arduino library for STM32Cube framework. Are they compatible?

ademyankov, 2019-05-09:
Shoot! :) How would I know?! I saw STM32 in the supported platforms and I thought it would work for my board.

Ok, so I have FreeRTOS port from STM32 System Workbench project that works with my board. What is the correct way to use it with VSCode and platformio? Use it not as an external dependency (library) but as local lib? (put it into lib folder?)

ivankravets, 2019-05-10:
Yes, please put a source code of this port to `project/lib/STM32RTOS/src`.

ademyankov, 2019-05-10:
Ok, got it... thanks! I am new to this iot world and there are a lot of confusing things.

One more question though, if these works, does it make sense to create a platformio library w/ that .library.json file and host this port on GitHub and publish it?

ivankravets, 2019-05-10:
>  does it make sense to create a platformio library w/ that .library.json file and host this port on GitHub and publish it?

Could you create a repo?

ademyankov, 2019-05-10:
I'll do it if I make it work with VS Code

ivankravets, 2019-05-23:
Hey all,

Does the latest version of PIO Core 4.0 work for you? If yes, we can close this issue. 

```
pio upgrade --dev
```

Thanks!

closed 2019-05-24 by ivankravets