platformio/platformio-core #1348: Home: Libraries: Stats

mmta41, 2018-02-09:
PIO Core Call Error: Error: You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.

mmta41, 2018-02-09:
updated vscode to latest version and there is no problem any more.
I'm not sure it was a vscode problem or not ,  thanks !

closed 2018-02-09 by ivankravets

kaunix, 2022-03-04:
I am having this issue. no proxy server and the firewall is set to allow vscode all connections... I've even tried it with the firewall off just to test and got the same error:
Error: you are not connected to the Internet. PlatformIO needs the Internet connection to download dependent packages or to work with the PlatformIO Account

ttww, 2022-03-11:
Also having this problem, no proxy, but Sophos firewall, but it does not block platformio.org

ttww, 2022-03-12:
> Also having this problem, no proxy, but Sophos firewall, but it does not block platformio.org

**Solution/Reason:**
 Looks like it was an older project and there were breaking changes in platform.ini an library references:

I had to change

`lib_deps=
    U8g2 @ 2.32.10
`
to
`lib_deps=
    olikraus/U8g2 @ 2.32.10
`

:-) :-(
A simple warning for this changes during compilation would be good....
However, I understand the motivation behind this change and thanks for this great project ;-)

LeHungryBoi, 2023-10-23:
this happened to me, and I switched from socks5 to http and it worked.

dhunganam161, 2023-10-28:
> this happened to me, and I switched from socks5 proxy to http proxy and it worked.

How did you do that?