platformio/platformio-core #1364: attachInterrupt function arduino

dvera2004, 2018-02-14:
For this:
  attachInterrupt (digitalPinToInterrupt(2), myFuction, FALLING);

The compiler return:
src/main.cpp:25:45: error: 'myFunction' was not declared in this scope
attachInterrupt(digitalPinToInterrupt(2), myFunction, FALLING);

I have defined myFunction correctly.

Then same code in arduino IDE compile without any problem.

Thanks

ivankravets, 2018-02-14:
attachInterrupt(digitalPinToInterrupt(2), &myFunction, FALLING);

closed 2018-02-14 by ivankravets

dvera2004, 2018-02-15:
Ivan,

thanks for your fast support.

I added the "&" but the issue still in there:

src/main.cpp:25:46: error: 'myFunction' was not declared in this scope
attachInterrupt(digitalPinToInterrupt(2), &myFunction, FALLING);
^
*** [.pioenvs/nanoatmega328/src/main.o] Error 1

Regards

ivankravets, 2018-02-15:
Could you share a simple INO files to reproduce this issue?

dvera2004, 2018-02-15:
Yes, 
[main.cpp.zip](https://github.com/platformio/platformio-core/files/1727559/main.cpp.zip)

dvera2004, 2018-02-15:
Ivan

I found the problem.
I'm using the default name for the source file, main.cpp (suggested by software), and the compiler maybe check the extension of file for compile. I renamed to main.ino and compiled without any problem.
When a new project is started and selecting an arduino board, atom must suggest a main.INO by default.

Thanks

ivankravets, 2018-02-15:
Ah... INO format is not a valid C/C++ file. It should die. People think that code in C but "INO Preprocessor" converts their NONAME format to valid C/C++ on-the-fly. As result, you can have a lot of issues.
See http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

dvera2004, 2018-02-15:
Ivan thanks for your kindly support.
I'm not understanding well about source code.

In arduino IDE every time I put all necessary "#include <Arduino.h>", compile and works, and the final version of .ino looks like a converted by hand ino file to c++.

The same code in Atom, not work like a c++ extension, but look like converted to c++ file, but when I only renamed the extension of the file and compile without any problem and the board works too.

ivankravets, 2018-02-15:
Indeed, you can use that file without modifications with PIO too. Just rename it from CPP to INO.

But... Advanced features, such as code linting and code completion will not work because INO is not CPP file.

dvera2004, 2018-02-15:
OK.

OK.

I will write all code using a .cpp file for have "code completion" service and in the last moment I will change to INO if is needed.

Thanks again.

zalexzperez, 2018-05-29:
Is there really no fix for this? I just go the same error, changing the extension from .cpp to .ino allowed me to compile the code successfully.

ivankravets, 2018-05-29:
@zalexzperez just add `&` before callback

zalexzperez, 2018-05-29:
@ivankravets Well, I already tried that and also writing the myFunction() on top of void setup but I get the same error...

ivankravets, 2018-05-29:
please provide a simple code to reproduce this issue. Also, be careful with spaces before function line. New line should start with a RETURN type of a function

ivankravets, 2018-05-29:
@zalexzperez I see. You can't use INO source code as C++ without modification. INO is not valid C++ code, you need to convert your project from INO to CPP manually. See Please read http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

zalexzperez, 2018-05-29:
@ivankravets Hmm... that sounds like a hassle. I'm sorry, but I'm new to the IDE world.

Anyway, what do I lose by changing the file extension from .cpp to .ino ? I see everything is working perfectly after doing so. Code completion still works (for instance, I write analogWr and the full command suggestion appears)

ivankravets, 2018-05-29:
> I'm sorry, but I'm new to the IDE world

You can stay with INO in this case (Arduino IDE compatible mode).

mattmunee, 2020-03-27:
> @ivankravets Okay. But right now I'm struggling to add external libraries after changing the extension to .ino.
> I pasted the library source code in a subfolder under /lib folder, then use the command #include "FastLED.h" and I'm getting the following error:
> 
> #include errors detected. Please update your includePath. IntelliSense features for this translation unit (C:\Users\Alejandro\Documents\PlatformIO\Projects\Prueba004\src\main.ino) will be provided by the Tag Parser.
> cannot open source file "FastLED.h"
> [image]

Have you fixed this issue?  I'm struggling to get files in my "lib" folder included.  Platformio doesn't seem to recognize them.  The README makes me think I can just put source into the "lib" folder and include it in my project main.cpp with no issue.  That doesn't seem to be the case. (PlatformIO version 4.3.2a1)

ivankravets, 2020-03-27:
1. I'm not sure that VSCode's C/C++ extension support INO format. INO is not a CPP file and you should avoid using INO files. See https://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually