platformio/platformio-core #1366: Platform espressif8266: Problem switching frameworks

freedib, 2018-02-15:
Trying to port an application from Arduino to RTOS, I alternate between _arduinoespressif8266_ and  _esp8266-rtos-sdk_ frameworks on my d1_mini.
When _esptool_ is used with the _arduinoespressif8266_ framework, it clears all the flash memory including the data located at the end of flash memory.
RTOS expects to have _esp_init_data_default.bin_ at the end of flash memory (address 0x3fc000 with 4Mb flash size) and won't boot if cleared.
The framework don't upload this file when uploading _eagle.flash.bin_ and _eagle.irom0text.bin_ considering it is static data and won't be erased.
Is there a way in platformio to prevent _arduinoespressif8266_ framework to erase the end of flash memory or in _esp8266-rtos-sdk_ framework to systematically _upload esp_init_data_default.bin_ with the application?

ivankravets, 2018-02-15:
Please move to appropriate repository and close this issue

freedib, 2018-02-15:
This issue was moved to platformio/platform-espressif8266#71

closed 2018-02-15 by freedib