platformio/platformio-core #1367: Custom platform packages per build environment

ivankravets, 2018-02-15:
Would be good to have an option which will allow overriding built-in dev/platform packages or providing a new one. Proposition is to add `platform_packages` option to `platformio.ini`. For example,

```
[env:myenv]
platform = ...
framework = ...
board = ...
platform_packages =
  pkg_with_spec_version@<4.5
  git_pkg@http://github.com/...
```

boneskull, 2018-02-15:
Ivan, if I was going to implement, where would I start?

boneskull, 2018-02-15:
Insofar as "custom tooling", I'm seeing the [Custom Uploader](http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#id3) docs, but they don't make a distinction between "uploading firmware" and "uploading something else".  In other words, `pio run -t upload` vs. `pio run -t uploadfs`.  

In my [original post](https://community.platformio.org/t/per-project-tools/3431/2), the tool mentioned is for uploading a filesystem.  It's also worth mentioning that I need *two separate tools* to upload *two separate filesystems*.

ivankravets, 2018-02-17:
In your case, you  need a custom target

* http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-build-target

alexwhittemore, 2019-03-19:
Is it safe to bet based on the lack of documentation that no such functionality has made it to the point of release yet? For example, I'm trying to add support for a custom SAMD board, using the Arduino framework. As far as I can tell, I'll have to fork [platformio-pkg-framework-arduinosam](https://github.com/platformio/platformio-pkg-framework-arduinosam) and add a custom arduino "variant" for my board specifics, then also fork [platform-atmelsam](https://github.com/platformio/platform-atmelsam) and add a "board.json" describing the board to PlatformIO. Then in `platformio.ini` I can add the repository URL of the custom platform.

Is all of that still necessary, as of now?

ivankravets, 2019-03-20:
@alexwhittemore you can put "variant" folder to ~/.platformio/packages/framework-atmelsam

As for the "board.json", please create "boards" folder in a project and put your "board.json" here.

alexwhittemore, 2019-03-20:
If I do that, however, it'll get clobbered by mainline package updates, no? And even if the update mechanism doesn't clear out untracked changes, that method seems like it's got low portability for replicating environments across multiple developers. That vs, say, having a "packages/framework-atmelsam/..." directory inside the project that gets considered, but tracked under the project's git repo.

Maybe there's not a super duper clean way to accomplish this and that's why it's not included yet. But if board.json can be covered by existing mechanisms, maybe doing something similar for the framework side is reasonable. And either way, I'd LOVE to see official documentation on the best way to accomplish this as of the current most recent release, since it's pretty opaque as-is.

ivankravets, 2019-03-23:
@alexwhittemore do you have any ideas on how to better resolve this issue?

alexwhittemore, 2019-03-23:
Not exactly, no, but I do have ideas for documentation.

My perfect situation would be to add a ‘boards/myboard.json’ and a ‘variants/myvariant’ dir, but there are a bunch of reasons that it’d be a bad hack at best. How would PIO know which framework it was for? Is the directory structure the same for frameworks besides Arduino? Etc. 

I think the “correct” solution is to add documentation about how to do just this, for example, for the Arduino framework. For instance, searching for “add a board” led me on a long journey to finally figure out that, for example, “framework version” can be a full url to a git repo. 

I’m happy to have a go at writing that up if there’s some kind of infrastructure for documentation PRs? It’ll probably require a bit of post processing given I’m far from a PIO expert.

closed 2019-06-07 by ivankravets

ivankravets, 2019-06-07:
See docs https://docs.platformio.org/en/latest/projectconf/section_env_general.html

alexwhittemore, 2019-06-21:
This is really excellent, thanks!

commit 68a3b3f9 (2019-06-07): Custom platform_packages per a build environment with an option to override default // Resolve #1367