platformio/platformio-core #1380: Fails to update (did not even tried to update it started automatically and failed)

benchuk, 2018-02-20:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
Please wait while upgrading PlatformIO...
Updating contrib-pysite                  @ 0.1.4          [0.1.5]
Uninstalling contrib-pysite @ 0.1.4:    Please manually remove file `C:\Users\Ben\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings\_constant_time.pyd`
Error: Traceback (most recent call last):
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\ben\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\__main__.py", line 74, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 44, in on_platformio_start
    after_upgrade(ctx)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 192, in after_upgrade
    update_core_packages(silent=True)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 95, in update_core_packages
    pm.update(name, requirements, only_check=only_check)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 56, in update
    result = PackageManager.update(self, *args, **kwargs)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 772, in update
    self.uninstall(pkg_dir, trigger_event=False)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 713, in uninstall
    util.rmtree_(pkg_dir)
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\util.py", line 774, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Users\Ben\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\Ben\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\Ben\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\Ben\.platformio\python27\Lib\shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "c:\users\ben\.platformio\penv\lib\site-packages\platformio\util.py", line 772, in _onerror
    raise e
WindowsError: [Error 5] Access is denied: 'C:\\Users\\Ben\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings\\_constant_time.pyd'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2018-02-20:
Please remove `C:\Users\Ben\.platformio\packages\contrib-pysite` and restart IDE

closed 2018-02-20 by ivankravets

benchuk, 2018-02-21:
FYI : Windows was holding up to the files had to restart the computer