platformio/platformio-core #1381: MacOs High Sierra: Could not load mDNS services

realjax, 2018-02-21:
**Operating system**: Macos 10.13.3

**PlatformIO Version** Home 0.6.1·Core 3.5.2b5

### Description of problem

Select devices -> Multicast DNS 

### Actual Results

Could not load mDNS services

### Expected Results

al list of devices

### Additional info

The output of the terminal command : 
_pio device list --mdns --json-output_

is this:

Error: Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/platformio/commands/device.py", line 50, in device_list
    return click.echo(json.dumps(data[single_key] if single_key else data))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 243, in dumps
    return _default_encoder.encode(obj)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py", line 207, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py", line 270, in iterencode
    return _iterencode(o, 0)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa4 in position 0: invalid start byte

============================================================

ivankravets, 2018-02-21:
Please provide an output of `pio device list --mdns` in Terminal

realjax, 2018-02-22:
?? I already did, see above.

ivankravets, 2018-02-22:
Without JSON option

realjax, 2018-02-23:
ok, here you go:

**************************************************************************************************************************
Obsolete PIO Core v3.5.1 is used (previous was 3.5.2b5)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
**************************************************************************************************************************
macbook._companion-link._tcp.local.
--------------------------------------
Type: _companion-link._tcp.local.
IP: 192.168.1.221
Port: 58514
Properties: rpVr=120.51; rpBA=5B:14:72:28:D7:3C

macbook._smb._tcp.local.
---------------------------
Type: _smb._tcp.local.
IP: 192.168.1.221
Port: 445

macbook._nfs._tcp.local.
---------------------------
Type: _nfs._tcp.local.
IP: 192.168.1.221
Port: 2049

MacBook Pro._companion-link._tcp.local.
----------------------------------------------
Type: _companion-link._tcp.local.
IP: 192.168.1.231
Port: 55007
Properties: rpVr=120.51; rpBA=49:C7:3F:24:1B:57

Chromecast-c74032dc260a2e6530af1a39e93f6c89._googlecast._tcp.local.
-------------------------------------------------------------------
Type: _googlecast._tcp.local.
IP: 192.168.1.191
Port: 8009
Properties: md=Chromecast; ve=05; rs=False; ca=4101; nf=1; cd=F104BBBAB6700319DF3747792A1798F3; bs=FA8FCA70A2F1; rm=False; ic=/setup/icon.png; st=0; id=c74032dc260a2e6530af1a39e93f6c89; fn=Woonkamer

Tr1kaIbYVn2ZVZhVCJyMBQ@8Y85Tkr3XS2iXMzyg3G02A._acp-sync._tcp.local.
-------------------------------------------------------------------
Type: _acp-sync._tcp.local.
IP: 169.254.11.211
Port: 5009
Properties: mac=DC:A4:CA:EA:71:6F; bssinfo=AQJkgIYQAAAJ3KTK7agDAVGGCAAAB9ykyu2oAg; nm=AirPort Time Capsule; cu=4dde65b7-ee90-555f-a5a9-f20ee14e9322; ssid=TuiPrint

AirPort Time Capsule._airport._tcp.local.
-----------------------------------------
Type: _airport._tcp.local.
IP: 169.254.11.211
Port: 5009
Properties: waMA=DC-A4-CA-EA-71-6F,raMA=DC-A4-CA-ED-A8-03,raM2=DC-A4-CA-ED-A8-02,raNm=TuiPrint,raCh=100,rCh2=1,raSt=0,raNA=0,syFl=0x8A0C,syAP=119,syVs=7.7.9,srcv=77900.2,bjSd=28

AirPort Time Capsule._afpovertcp._tcp.local.
--------------------------------------------
Type: _afpovertcp._tcp.local.
IP: 169.254.11.211
Port: 548

AirPort Time Capsule._adisk._tcp.local.
---------------------------------------
Type: _adisk._tcp.local.
IP: 169.254.11.211
Port: 9
Properties: sys=waMA=DC:A4:CA:EA:71:6F,adVF=0x1010; dk2=adVF=0x1093,adVN=Data,adVU=24c6f2a3-17d5-5fcf-9340-bc59dff5fdf0

Jenkins._http._tcp.local.
-------------------------
Type: _http._tcp.local.
IP: 192.168.1.40
Port: 5000
Properties: url=http://192.168.1.40:5000/; path=/; version=1.647; server-id=2b1ed33bfc54cfd69d2aaf5627780c5c; slave-port=1047

50-34-10-70.1 AirPort Time Capsule._sleep-proxy._udp.local.
-----------------------------------------------------------
Type: _sleep-proxy._udp.local.
IP: 169.254.11.211
Port: 50522

jenkins._jenkins._tcp.local.
----------------------------
Type: _jenkins._tcp.local.
IP: 192.168.1.40
Port: 5000
Properties: url=http://192.168.1.40:5000/; path=/; version=1.647; server-id=2b1ed33bfc54cfd69d2aaf5627780c5c; slave-port=1047

jenkins._hudson._tcp.local.
---------------------------
Type: _hudson._tcp.local.
IP: 192.168.1.40
Port: 5000
Properties: url=http://192.168.1.40:5000/; path=/; version=1.647; server-id=2b1ed33bfc54cfd69d2aaf5627780c5c; slave-port=1047

anMCU._arduino._tcp.local.
--------------------------------
Type: _arduino._tcp.local.
IP: 192.168.1.245
Port: 8266
Properties: tcp_check=no; ssh_upload=no; board=ESP8266_NODEMCU; auth_upload=yes

Officejet 6500 E710n-z [E7FCC2]._ipp._tcp.local.
------------------------------------------------
Type: _ipp._tcp.local.
IP: 192.168.1.45
Port: 631
Properties: note=False; URF=CP1,MT1-2-8-9-10-11,OB9,OFU0,PQ3-4-5,RS300-600,SRGB24,W8,DM3,IS1; rp=ipp/printer; ty=Officejet 6500 E710n-z; adminurl=http://HPE7FCC2.local.; Duplex=T; qtotal=1; product=(HP Officejet 6500 E710n-z); Scan=T; priority=60; Color=T; pdl=application/vnd.hp-PCL,image/urf,image/jpeg; txtvers=1

Officejet 6500 E710n-z [E7FCC2]._pdl-datastream._tcp.local.
-----------------------------------------------------------
Type: _pdl-datastream._tcp.local.
IP: 192.168.1.45
Port: 9100
Properties: note=False; product=(HP Officejet 6500 E710n-z); ty=Officejet 6500 E710n-z; adminurl=http://HPE7FCC2.local.; Duplex=T; qtotal=1; Scan=T; priority=30; Color=T; pdl=application/vnd.hp-PCL,image/urf,image/jpeg; txtvers=1

pi-audio [b8:27:eb:20:14:3d]._workstation._tcp.local.
-----------------------------------------------------
Type: _workstation._tcp.local.
IP: 192.168.1.143
Port: 9

pi-audio._afpovertcp._tcp.local.
--------------------------------
Type: _afpovertcp._tcp.local.
IP: 192.168.1.143
Port: 548

realjax, 2018-03-01:
Anyone?  I tried what is mentioned here:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system

But it made no difference. Still same problem

ivankravets, 2018-03-08:
I see that you have multiple PIO Cores in a system. Please open SYSTEM terminal (not PIO IDE) and run `sudo pip uninstall platformio`

realjax, 2018-03-08:
Already tried that. Doesn't make a difference.

ivankravets, 2018-03-08:
Please unpack the attached archive, open PIO IDE Terminal, and type
```
python /path/to/test_mdns.py.zip
```
Please paste an output on https://hastebin.com

[test_mdns.py.zip](https://github.com/platformio/platformio-core/files/1794234/test_mdns.py.zip)

realjax, 2018-03-11:
Thanks, out put is not much, but here it is:
https://hastebin.com/zuravoseri.sql

ivankravets, 2018-03-12:
You NEED to use PIO IDE Terminal (terminal icon on toolbar), not system terminal.

realjax, 2018-03-12:
I DID.

realjax, 2018-03-23:
Ok.

Seems to be caused by an Airport time capsule in my network.  It displays a weird netbios name 👍 AirPort._smb._tcp.local.
------------------------
Type: _smb._tcp.local.
IP: 169.254.11.211
Port: 445
Properties: netbios=��r

And multicast DNS chokes on that netbios name. When I switch off the airport, then multicast DNS works fine.
I cannot alter the netbios name of the Airport ( of course not, its Apple) so I'm kinda stuck here.

realjax, 2018-03-23:
Thanks, I tried and It said:
PlatformIO has been successfully upgraded to 3.5.3a7

But the problem remains after the restart :(

ivankravets, 2018-03-23:
Could you provide TeamViewer access to debug this issue? Please send me ID/Pass to me@ikravets.com

ivankravets, 2018-03-23:
Could you try again `pio upgrade --dev` and restart IDE? I did some changes to the code.

realjax, 2018-03-24:
Succes!!  This worked !  Thanks a bunch Ivan :)

ivankravets, 2018-03-24:
Thank you so much for the report! ;)

closed 2018-03-24 by ivankravets