platformio/platformio-core #1383: Feature Request: Simultaneously upload to multiple serial ports

cjstott94, 2018-02-22:
Feature Request.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10
**PlatformIO Version** 3.5.2b5

### Description of problem
Have multiple ESP32s that will communicate with each other over wifi/LoRa. 
Would be handly to upload new firmware changes to all connected devices at once (ie. multiple instances of esptool, preferably with an auto retry counter as it sometimes fails). 

Currently, it will only upload to the first found device (or device on configured port). Meaning to update all devices requires either specifying a port each upload or disconnecting and connecting next device which is very time-consuming.

Thought PIO remote might be able to do this, though it seems the agent does the same thing. 

**The content of `platformio.ini`:**
```ini
[env:heltec_wifi_lora_32]
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
monitor_baud = 115200
upload_speed = 921600
```

ivankravets, 2018-02-22:
You can automate this procedure with http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

cjstott94, 2018-02-23:
Thanks for that.
What I managed to come up with for now.

seemed to work best doing it after 'size' build target
doing it after upload target still meant waiting for two upload cycles and not sure how to make a custom build target that actually rebuilds the project before uploading

**Content of `platformio.ini`:**
```ini
[env:heltec_wifi_lora_32]
extra_scripts = multi_esptool.py
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
; upload_port = COM3
simultaneous_upload_ports =  COM3, COM6, COM16
upload_speed = 921600
```
**Content of `multi_esptool.py`:**
```py
Import('env')
import threading
from threading import Thread
from base64 import b64decode

print "Current build targets", map(str, BUILD_TARGETS)
returnCodes=[]#list of tuples storing com port and upload result
def run(port):
    env.Replace(UPLOAD_PORT=port)
    command = env.subst('$UPLOADCMD') +" "+ env.subst('$BUILD_DIR\$PROGNAME') +".bin"
    for i in range (0,3):#try up to 3 times
        errorCode=env.Execute(command)
        if (errorCode==0):
            returnCodes.append((port,errorCode))
            return
    returnCodes.append((port,errorCode))

def after_build(source, target, env):
    simultaneous_upload_ports = ARGUMENTS.get("SIMULTANEOUS_UPLOAD_PORTS")
    if (simultaneous_upload_ports!=None):#check if defined
        threads = []
        ports = map(str.strip, b64decode(simultaneous_upload_ports).split(','))
        print "Uploading to " + str(ports)
        for port in ports :
            thread = Thread(target=run, args=(port,))
            thread.start()
            threads.append(thread)
        for thread in threads:
            thread.join()#wait for all threads to finish
        encounteredError=False
        for code in returnCodes:
            if(code[1]==0):
                print code[0] + " Uploaded Successfully"
            elif(code[1]==1):
                print code[0] + " Encountered Exception, Check serial port"
                encounteredError=True
            elif(code[1]==2):
                print code[0] + " Encountered Fatal Error"
                encounteredError=True
        if(encounteredError):
            Exit(1)

env.AddPostAction("size", after_build)#"buildprog" dowesnt seem to work
# env.AddPostAction("upload", after_build)#uploads to port specified by upload_port first then does simultaneous_upload_ports
# AlwaysBuild(env.Alias("uploadAll", "", after_build))#not sure how to make custom target to build first

ivankravets, 2018-03-08:
Please use our implementation instead of your `def serial_ports():`
```
from platformio import util

util.get_serial_ports()
```

We will add this example to our docs later.

closed 2018-03-08 by ivankravets

parsley, 2024-09-30:
Searching the internet for a solution for simultaneous upload to multiple devices this (closed) FR was still the best hint I clould find. Thanks @cjstott94 and @jacklu333333 for these templates.

Since on my machines the script from @jacklu333333 complained about missing colored and pyudev I modified it further and also fixed the issue he had with passing the variable `simultaneous_upload_ports`. I can now pass either AUTO or comma separated ports like COM1,COM3, COM23 or /dev/cu.usbserial70,/dev/cu.usbserial90 or /dev/ttyUSB0 from platformio.ini to the script.

This version works for me on both systems I currently use:
- Windows 10 with VScode and Platformio extension
- MacOS Sonoma 14.7 with VScodium and Platformio extenstion

On Windows the script filters detected COM ports for "USB VID:PID=1A86:" and on macOS for "/dev/cu.usbserial" to avoid unnecessary error messages.
I did not try it on Linux but I think it should be easy to use it and add similar filtering.

Good luck and have fun with nice projects.

```python
Import("env")
import glob
import sys
import threading
from base64 import b64decode
from threading import Thread

import serial
from platformio import util

print("\nextra script info:")
print("  Monitor speed:", env.GetProjectOption("monitor_speed"))
print("  Simultaneous upload ports:", env.GetProjectOption("simultaneous_upload_ports"))
print("  Current build targets:", list(map(str, BUILD_TARGETS)))

returnCodes = []  # list of tuples storing com port and upload result

def run(port):
    env.Replace(UPLOAD_PORT=port)
    for i in range(0, 3):  # try up to 3 times
        if i > 0:
            env.Replace(UPLOAD_SPEED="115200")  # try slowing down baud
        command = (
            env.subst("$UPLOADCMD")
            + " "
            + env.subst("$BUILD_DIR")
            + "/"
            + env.subst("$PROGNAME")
            + ".bin"
        )

        errorCode = env.Execute(command)
        if errorCode == 0:
            returnCodes.append((port, errorCode))
            return

    returnCodes.append((port, errorCode))

def after_build(source, target, env):
    print("Build finished\n\nmulti_esptool.py script:")
    simultaneous_upload_ports = env.GetProjectOption("simultaneous_upload_ports")
    # print(f"Simultaneous upload ports: {simultaneous_upload_ports}")
    if simultaneous_upload_ports != None:  # check if defined
        threads = []
        ports = simultaneous_upload_ports
        print(f"Given port(s): {ports}")
        if str(ports) == "AUTO":
            print("getting serial ports...")
            ports = util.get_serial_ports()
            print("I got these ports:", ports)
            # Windows
            if "port': 'COM" in str(ports):
                print("-> looks like we're on a Windows system...")
                port_list = []
                for port in ports:
                    if "USB VID:PID=1A86:" in str(port["hwid"]):
                        port_list.append(port["port"])
                print("-> filtered ports for VID=1A86: " + str(port_list))
            else:
                # macOS
                if "/dev/cu.usbserial" in str(ports):
                    print("-> looks like we're on an Apple macOS...")
                    port_list = []
                    for port in ports:
                        if "/dev/cu.usbserial" in str(port):
                            port_list.append(port["port"])
                    print("-> filtered ports for /dev/cu.usbserial: " + str(port_list))
                # other OS
                else:
                    port_list = []
                    for port in ports:
                        port_list.append(port["port"])
                    print("-> I found these ports:", port_list)
        else:
            # split configuration string from platformio.ini into a list of ports
            port_list = str(ports).split(",")
        print("I'll try uploading to ", port_list)
        for port in port_list:
            # thread = Thread(target=run, args=(port,))
            thread = Thread(target=run, args=(str(port).replace(" ", ""),)) # remove blanks
            thread.start()
            threads.append(thread)
        for thread in threads:
            thread.join()  # wait for all threads to finish
        encounteredError = False
        returnCodes.sort(key=lambda code: code[0])
        for code in returnCodes:
            if code[1] == 0:
                print(f"{code[0]} Uploaded Successfully")
            elif code[1] == 1:
                print(f"{code[0]}: Encountered Exception, Check serial port")
                encounteredError = True
            elif code[1] == 2:
                print(f"{code[0]}: Encountered Fatal Error")
                encounteredError = True
        if encounteredError:
            Exit(1)
        Exit(0)
    else:
        print("No Simultaneous Upload Ports Defined")

env.AddPreAction("upload", after_build)
```

Towerful, 2024-10-06:
Anyone on Windows using ESP32 trying this, any spaces in folders will break this.  
Platformio seems to work fine when its used normally, but this breaks the `<address> <path>` sequence expected by `esptool`.  
It seems like a known bug, as there have been a few issues ( eg https://github.com/platformio/platform-espressif32/issues/118 and https://github.com/platformio/platform-espressif32/issues/121 ).  

I had a tinker with the builder script (referenced in one of those issues). I can't seem to chase out the broken paths.  

I ended up moving my project & platformio install to some root folders, so there are no space.  
Its probably a documented requirement for platformio, but I one-click installed via VSCode so it passed over my head.