platformio/platformio-core #1386: Home: Could not install platform

MacPit69, 2018-02-23:
PIO Core Call Error: PlatformManager: Installing espressif32
Warning! Package Mirror: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/platformio/platform-espressif32/tarball/v0.12.0 (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))
Looking for another mirror...
Warning! Package Mirror: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/platformio/platform-espressif32/tarball/v0.12.0 (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))
Looking for another mirror...

Error: Could not install 'espressif32' with version requirements '*' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

ivankravets, 2018-02-24:
Duplicate of https://github.com/platformio/platformio-core/issues/1390

closed 2018-02-24 by ivankravets

ivankravets, 2018-02-24:
Could you re-test now?