platformio/platformio-core #1390: Home: Could not install platform

r100gs, 2018-02-24:
PIO Core Call Error: PlatformManager: Installing espressif32
Warning! Package Mirror: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/platformio/platform-espressif32/tarball/v0.12.0 (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))
Looking for another mirror...
Warning! Package Mirror: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/platformio/platform-espressif32/tarball/v0.12.0 (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))
Looking for another mirror...

Error: Could not install 'espressif32' with version requirements '*' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

zerog2k, 2018-02-24:
github disabled some weaker SSL options yesterday - broke lots of folks...
I've seen this on older curl/openssl, and updating openssl and curl helped.
I'm having this on mac... currently trying to figure out what will take to fix.

https://github.com/blog/2507-weak-cryptographic-standards-removed

zerog2k, 2018-02-24:
@ivankravets looks like github doesnt only supports TLS 1.2 now, which I cant figure out how to make mac built-in python do it. (sierra 10.12.6 = python 2.7.10 which seems to be linked to old openssl:
>>> import ssl
>>> ssl.OPENSSL_VERSION
'OpenSSL 0.9.8zh 14 Jan 2016'

i have newer python3 installed, but not sure if pio is using it's own python installation, or if it's possible to point it to another, or if there is some other solution here...

zerog2k, 2018-02-24:
Ok, I see you are using a virtualenv
(penv) hackpro:~ jens$ which python
/Users/jens/.platformio/penv/bin/python
(penv) hackpro:~ jens$ python
Python 2.7.10 (default, Feb  7 2017, 00:08:15) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.34)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssl
>>> ssl.OPENSSL_VERSION
'OpenSSL 0.9.8zh 14 Jan 2016'
>>> 

Is it correct that this will not be able to support TLSv1.2?

ivankravets, 2018-02-24:
I've just created a mirror of packages in our PlatformIO DL Storage. Please re-test again.
It should fail with Github but the next PlatformIO mirror should work.

closed 2018-02-24 by ivankravets