platformio/platformio-core #1401: Kdevelop integration Feature Request

vayulove, 2018-02-26:
What kind of issue is this?

- [x] Feature Request. 

------------------------------------------------------------------

**Operating system**:  debian sid - KDE

**PlatformIO Version** (`platformio --version`):  version 3.5.2rc1

### Description of problem
KDE is a premier desktop environment with integration between KDE applications and numerous features that aide in programming workflow. When using non KDE applications there is often a feeling quality that something's missing. Kate/Kdevelop are integral to that environment.  Kdevelop's simplicity makes it a powerful tool that doesn't get in the way of your process.

#### Steps to Reproduce

I can run platformio init from within the folder of my project. Then within Kdevelop I can configure a launch to execute "platformio run -t upload" and it works to compile and upload but there needs to be a cmake file in order to have my project display external library symbols.

I tried to use the one from using "pio init --ide clion" but it compiles with errors and it fails to create the symbols to use within the editor.

ivankravets, 2018-03-06:
Could you provide configuration files for Kdevelop? See https://github.com/platformio/platformio-core/tree/develop/platformio/ide/tpls

vayulove, 2018-03-15:
KDevelop needs cmake files similar to clion. Unfortunately I have tried over and over and haven't been able to figure out cmake enough to make it work.

vayulove, 2018-03-15:
I'm not sure what you need.  I'll help with anything I can understand (unfortunately that's not much, but I'll be happy to do anything you can describe step by step).  I'm attaching here a C++ hello world project made with KDevelop.  I hope that might help.
[helloworld.tar.gz](https://github.com/platformio/platformio-core/files/1817116/helloworld.tar.gz)

Tech1207, 2025-09-07:
Has a solution to this been made/found? I know it's like 7 years later but I'm sick of copilot and windows stuff.