platformio/platformio-core #1404: avr-g++: error: No such file or directory

albe2669, 2018-02-27:
What kind of issue is this?

- [ ] PlatformIO IDE.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

Gonna be honest don't know if this is the IDE or the Core
------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10: Home, Creators update

**PlatformIO Version**:
3.5.1
### Description of problem
When i try to upload/build i get a lengthy error starting with "avr-g++: error: src\main.cpp: No such file or directory" which looks like is what causes the rest of the error message

#### Steps to Reproduce

1. Install atom
2. Install python
3. Install platformio-core
4. Install platformio-ide
5. Make a new project
6. Write the blink example
7. Build

### Actual Results
```
[02/27/18 11:57:37] Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 24 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\uno\src\main.o
Archiving .pioenvs\uno\libFrameworkArduinoVariant.a
Compiling .pioenvs\uno\FrameworkArduino\CDC.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial.o
avr-g++: error: src\main.cpp: No such file or directory
avr-g++: fatal error: no input files
compilation terminated.
c:\users\alber\.platformio\packages\toolchain-atmelavr\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: .pioenvs\uno\libFrameworkArduinoVariant.a: No such file or directory
*** [.pioenvs\uno\src\main.o] Error 1
*** [.pioenvs\uno\libFrameworkArduinoVariant.a] Error 1
Assembler messages:
Fatal error: can't create .pioenvs\uno\FrameworkArduino\CDC.o: No such file or directory
*** [.pioenvs\uno\FrameworkArduino\CDC.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs\uno\FrameworkArduino\HardwareSerial.o: No such file or directory
*** [.pioenvs\uno\FrameworkArduino\HardwareSerial.o] Error 1
============================================== [ERROR] Took 1.55 seconds ==============================================
```

### Expected Results
A succesful build with no errors

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include "Arduino.h"

#ifndef LED_BUILTIN
#define LED_BUILTIN 13
#endif

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(1000);
}

```

ivankravets, 2018-03-06:
Try to remove `c:\users\alber\.platformio\packages\toolchain-atmelavr` folder and restart build

closed 2018-03-06 by ivankravets

albe2669, 2018-03-07:
@ivankravets 
Hi.
Thanks for the answer, but sadly that didnt work.
I get exactly the same error after deleting the folder.
I also tried to completely uninstall atom, platformio and python, then reinstalling all of it again and i still got that error.
Do you have something else i can try?
Thanks

ivankravets, 2018-03-08:
Do you have Antivirus on a system?

albe2669, 2018-03-08:
Yes, i tried turning it off before installation but sadly no difference

snikul, 2020-12-12:
I tried to turn off my Windows Defender Antivirus but no good result.

clicurt, 2021-03-25:
Open Windows Defender → Virus & threat protection → Manage settings → Manage Controlled folder access → Allow an app through controlled folder access → Add an allowed app → Recently blocked apps → Choose app to allow.
You might need to do this several times until build is successful.