platformio/platformio-core #1407: Only the libraries used by the current configuration should be compiled

VioletGiraffe, 2018-02-28:
I have an Arduino project with multiple configurations. Different configurations use different libraries, and the problem is that PIO compiles everything it can find in .piolibdeps.  It means that if I have a library that's only compatible with some of the configurations, the other configurations will fail to build until I manually clean the .piolibdeps folder.

Suggested solution: wipe .piolibdeps on first build after environment variables have changed. Or at least wipe the libraries specified as `git` repos and as references to the PIO library registry.

ivankravets, 2018-03-06:
Could you provide a simple project to reproduce this issue?

VioletGiraffe, 2018-03-06:
Sure. Check it out: https://github.com/VioletGiraffe/platformio-1407-demo
See travis.yml for the scenario that triggers the error. And here's what it looks like (scroll to the bottom):
https://travis-ci.org/VioletGiraffe/platformio-1407-demo

ivankravets, 2018-03-06:
Please read http://docs.platformio.org/en/latest/librarymanager/ldf.html#storage. 

You need http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

closed 2018-03-06 by ivankravets

VioletGiraffe, 2018-03-06:
Hm. Regarding `lib_ignore`: if "The correct value for this option is library name (not folder name)", what is the library name for a GitHub repo library?

And don't you think that _for the user-specified dependencies_, only those specified by the current environment should be visible?

ivankravets, 2018-03-06:
> what is the library name for a GitHub repo library?

You should see library name in build log. See https://travis-ci.org/VioletGiraffe/platformio-1407-demo/jobs/349887944#L689

Also, you can control library name with `<name>=repository`. See examples http://docs.platformio.org/en/latest/userguide/lib/cmd_install.html

> only those specified by the current environment should be visible?

By default, PlatformIO scans dependency automatically. If you would like to manage them manually, please turn off LDF http://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode

VioletGiraffe, 2018-03-06:
>You should see library name in build log. 
Thanks.

> If you would like to manage them manually, please turn off LDF
I got that idea from your previous link, but it brought with it a host of issues I don't feel like dealing with. For instance, I couldn't get it to find "SPI.h" (any hints?).
What I'm proposing is a fully automatic solution that does not require any manual labor from the programmer, and does not induce any issues.

ivankravets, 2018-03-06:
> For instance, I couldn't get it to find "SPI.h" (any hints?).

lib_deps = SPI