platformio/platformio-core #1408: Home: Could not load boards list

zayden, 2018-02-28:
PIO Core Call Error: Error: You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.

kemaldede, 2018-02-28:
I have internet connection but I can not load board lists and libraries

ivankravets, 2018-03-06:
What is your OS?

ivankravets, 2018-03-12:
Please reopen if you still need help

closed 2018-03-12 by ivankravets

abdulnafea-auf, 2018-05-05:
I have also the same problem but while trying to install a library, I am working on Windows 10 education 64-bit.

jeffminton, 2018-05-23:
I am having this same issue on linux using the vs code ide

ivankravets, 2018-05-24:
@jeffminton could you reproduce this issue again?

gerhardh, 2018-10-29:
I Have the same problem using VS Code on macOS Mojave. I ave a very good 100 Mb/s Internet connection. However this problem shows up every few weeks for a few days.

ivankravets, 2018-10-29:
When do you see this message? After which command?

gerhardh, 2018-11-01:
I got this message whenever I tried to list either boards or frameworks or libraries in PIO Home. 
Problem is solved. 
Reason was interference from Little Snitch.
Had to delete some Little Snitch rules and PIO work properly again.
Thanks