platformio/platformio-core #1412: Feature request: Specify maximum size of binary

someburner, 2018-03-04:
Right now it appears pio only checks size before uploading. For CI and OTA purposes, it would be really nice if one could specify size as a build parameter and to fail the build if it is too large. May as well add minimum feature too as a sanity check for similar purposes.

ivankravets, 2018-03-06:
See

* http://docs.platformio.org/en/latest/faq.html#firmware-memory-size-explanation-text-data-and-bss
* http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

You can create own POST action, call gcc-size, read data and analyze them.

someburner, 2018-03-06:
Seen it.. I cannot get it to execute within a virtual environment, but even if I could, it seems like a lot of code duplication and mess to have it in a project when it could be added as its own flag.

me21, 2018-03-19:
It can be done by providing custom linker script.

ivankravets, 2018-06-01:
Please use `pio run -t checkprogsize` for CI/CD

commit 20c1ce40 (2018-06-01): Check maximum allowed firmware size for uploading with "pio run --target checkprogsize" // Resolve #1412
closed 2018-06-01 by ivankravets