platformio/platformio-core #1419: PlatformIO Home not launching ... eternal spinners

M-VdM, 2018-03-06:
What kind of issue is this?
- [x] PlatformIO Core. If you’ve found a bug, please provide an information below
------------------------------------------------------------------

### Configuration
**Operating system**:  Mac OS X El Capitan _(10.11.6)_
**PlatformIO Version** (`platformio --version`):  _3.5.1_

### Description of problem: PlatformIO Home not launching ... eternal spinners
This kind of issue has been reported dozens of times already ... _on all platforms._
I tried multiple (clean) re-installs over the last 4 months _(OS, atom, platformio, ...)_, but nothing seems to help.

Finally, I installed platformio-core via pip ('sudo pip install platformio')

Now I get this: The "**pio home**" command produces this output + the well known "Loading" spinners on the PlatformIO Home page
```
PlatformIO Plus (https://pioplus.com) v0.14.1

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________
```

The "**pio home --port 888**" command produces this output / errors:

```
PlatformIO Plus (https://pioplus.com) v0.14.1

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:888
|__|____|_______________________

Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 567, in cmd_home.cli
  File "/Users/Mario/.platformio/packages/contrib-pysite/twisted/internet/posixbase.py", line 495, in listenTCP
    p.startListening()
  File "/Users/Mario/.platformio/packages/contrib-pysite/twisted/internet/tcp.py", line 980, in startListening
    raise CannotListenError(self.interface, self.port, le)
CannotListenError: Couldn't listen on 127.0.0.1:888: [Errno 13] Permission denied.
============================================================
```

### Additional info
**Mac OS X Firewall** = Disabled
$ **echo $PATH**
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
$ **clang --version**
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
$ **python --version**
Python 2.7.10
$ **pip --version**
pip 9.0.1 from /Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg (python 2.7)
$ **pip list**
```
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.10.2)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
bottle (0.12.13)
certifi (2018.1.18)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (9.0.1)
platformio (3.5.1)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (3.4)
python-dateutil (1.5)
pytz (2013.7)
requests (2.18.4)
scipy (0.13.0b1)
semantic-version (2.6.0)
setuptools (1.1.6)
six (1.4.1)
urllib3 (1.22)
xattr (0.6.4)
zope.interface (4.1.1)
```

ivankravets, 2018-03-06:
For debug purpose please use 8888 port in system terminal. Do you see any errors?

M-VdM, 2018-03-10:
Sorry for the late reply.
### Debug Test:
$ **pio home --port 8888**
```
PlatformIO Plus (https://pioplus.com) v0.14.4

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8888
|__|____|_______________________
```
Result: No errors - Just the usual PlatformIO Home page + **PSoD** _(PlatformIO Spinners of Death ;-)_
A "**pio upgrade**", bumping contrib-piohome @ 0.7.1 didn 't help either.

### Other commands:
**$ pio update**    --> OK 
**$ pio upgrade**   --> OK
**$ pio platform frameworks / list / search ... / show ... / update**   --> OK
**$ pio platform install teensy**   --> OK
**$ pio platform install arduino / wiringpi / simba / xyz**  --> **_NOK !_**
PlatformManager: Installing arduino
Error: Detected unknown package 'arduino'

So:  installing any framework other then 'teensy' is unsuccessful ...

ivankravets, 2018-03-10:
Please mail me@ikravets.com for the further steps and details.

ivankravets, 2018-03-22:
Have you resolved this issue?

Nanne118, 2018-03-24:
I am encountering the same issue, though I run Windows 7 64-bit for both the Atom and VSCode installs I have tried so far. VSCode appealed to me more (after evaluating the differences for PlatformIO, as well as reviews from fellow developers) but like so many I cannot get my PlatformIO (Pio) Home to run.

I have tried reinstalling VSCode, deleting (.penv) and subsequently installing the newest version of Python, installing Atom and deleting all packages within Visual Studio Code only to redownload and reinstall them, to no avail getting PlatformIO to run. The closest to seeing anything resembling PlatformIO IDE run is the little pop-up message urging me to 'reload (VSCode) now', but it gives me this message whilst the status in the bar below still lists ''Installing PlatformIO IDE''. However, waiting does not appear to foot any progress towards the installation (at least, none that I can see) and as such I have no other course of action then to press 'Reload Now' in the hopes that this will fix things.

However, upon being greeted by the new instance of VSCode nothing hints of PlatformIO IDE being installed save for the supposed 'installed' category within the VSCode extensions menu. No Pio Home icon shows up on my bar to access, and running the platformIO home command through the SHIFT+CTRL+P command results in the sad loading screen that never seems to end.

ivankravets, 2018-03-24:
@Nanne118 could you try to navigate to OS Control Panel > Installed software and uninstall all Pythons from a sytem? After that, please remove `%USEHOME%/.platformio` folder and restart IDE.

ivankravets, 2018-03-26:
Menu> Help > Developer Tools > Console.

Do you see any errors? Maybe, Antivirus/Firewall?

Nanne118, 2018-03-26:
The error it is giving me is the `Error: Could not install 'contrib-pysite' with version requirements '>=0.2.0,<2' for your system 'windows_amd64'.` error. As is recommended I followed solutions given by the FAQ:
1. Removing the folder did not solve it: unsurprisingly as this did not work in the previous attempts either
2. I had added VSCode to my AntiVirus (BitDefender) before, but this did not seem to work correctly (as it did not show up in the whitelist). After completely shutting down all modules of my AntiVirus, I however still seem to get the error that for that specific package (Contrib-pysite) the access is denied `Warning! Package Mirror: [Error 5] Access is denied` for the two mirrors. Oddly enough, the package tool-pioplus is installed without any deferrment.
3. This step relies on setting your CMD to the correct directory (at the location of ''platformio.ini''), but this file does not seem to exist in either the users/.platformio , users/.vscode or D:/vscode folders. 
4. Lastly, it recommends installing the missing package manually. This step I had actually good hopes for, as it is simply a missing package that I could not download for whatever reason. I tried to download it manually (with Cygwin) from the recommend download address (https://dl.bintray.com/platformio/dl-packages/contrib-pysite), but VSCode now tells me that this is a broken .json package, so there goes that idea.

ivankravets, 2018-06-18:
@Nanne118 have you resolved this issue? It should be fixed in the latest release of VSCode.

ivankravets, 2018-06-18:
Please reopen if you still need help

closed 2018-06-18 by ivankravets

MkLHX, 2018-12-07:
Hi, i have the same problem on VS code. I published the https://github.com/platformio/platformio-vscode-ide/issues/449
because i don't know if the problem from the core or from ide

MkLHX, 2018-12-10:
hi! i fix it with : 
```
pip --no-cache-dir install -U platformio
```

Read here https://github.com/platformio/platformio-core/issues/295

arielbernal, 2019-04-19:
and a year later... same issue.  Installed from scratch, both vscode and platformIO. Removed all python versions, including anaconda. Nothing works. 

$ pio home
CorePackageManager: Installing contrib-pysite @ ~2.27.0
Error: Detected unknown package 'contrib-pysite'

Tried @MkLHX  fix using pip, but it didn't help..

ivankravets, 2019-04-19:
@arielbernal  see https://github.com/platformio/platformio-core/issues/2341

lyhistory, 2020-01-01:
try to run vscode in admin mode, that solved my issue