platformio/platformio-core #1427: Home: Internal Store Exception

ilyary, 2018-03-08:
TypeError: Cannot read property 'name' of null
    at http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:44845
    at Array.map (native)
    at b (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:44716)
    at http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:45235
    at Array.map (native)
    at _ (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:45175)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:17:243549)
    at r (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:10153)
    at l (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:11523)
    at u (http://127.0.0.1:8010/0237bd675d712281a0ee.min.js:6:11641)

ivankravets, 2018-03-08:
Could you reproduce this issue again?

ilyary, 2018-03-09:
I've just installed atom and pio-ide.
The issue took place when I intended to create first project (NodeMCU,
Arduino). I pressed Create and got invite to send an issue. The project was
created anyway, and the ide went on working.

Another project was created without an issue. But I was prompted to update
the ide and accepted. So maybe I got a fix

Love platformio

Cheers,
Ilya

9 мар. 2018 г. 0:45 пользователь "Ivan Kravets" <notifications@github.com>
написал:

Could you reproduce this issue again?

ivankravets, 2018-03-09:
Thanks! Don’t hesitate to reopen if you will catch this issue again.

closed 2018-03-09 by ivankravets