platformio/platformio-core #1433: `include_dir` option in platformio.ini not recognized

Piroro-hs, 2018-03-10:
### Configuration

**Operating system**: Xubuntu 17.10 64bit

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 3.5.2rc4`

### Description of problem

The `include_dir` option in the `platformio.ini` is not recognizeed by PlatformIO core.

#### Steps to Reproduce

1. Create `platformio.ini` with following contents.
   ```ini
   [platformio]
   include_dir = Inc
   ```

2. Run `pio run`.

### Actual Results

``Warning! Ignore unknown `include_dir` option in `[platformio]` section``

### Expected Results

According to the [docs](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#include-dir), it configures a path to project’s headers files.

ivankravets, 2018-03-12:
Thanks! Please open PIO IDE Terminal and type `pio upgrade --dev`. Does it work now?

closed 2018-03-12 by ivankravets

Piroro-hs, 2018-03-13:
It works! Thank you.

commit f63a6d73 (2018-03-12): Add "include_dir" to known options list // Resolve #1433