platformio/platformio-core #1436: Home: Could not initialize project

dropmeaword, 2018-03-11:
PIO Core Call Error: The next files/directories have been created in /Users/zilog/Documents/PlatformIO/Projects/something
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Error: [Mon Mar 12 00:10:49 2018] Processing d1_mini_pro (platform: espressif8266; board: d1_mini_pro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'd1_mini_pro'
========================== [ERROR] Took 0.71 seconds ==========================

dropmeaword, 2018-03-11:
the `d1_mini_pro` platform worked just fine until the last update.

ivankravets, 2018-03-12:
Could you run in PIO IDE Terminal `pio update` command and restart IDE? Does it work now?

ivankravets, 2018-03-15:
Please reopen if you still need help

closed 2018-03-15 by ivankravets

bendem, 2018-09-29:
Having the same problem here:

```
➜  gate-weemos git:(master) ✗ pio update
Updating tool-scons                      @ 3.20501.2      ^O[Up-to-date]
Updating framework-arduinoavr            @ 1.10621.1      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]

Library Manager
===============
➜  gate-weemos git:(master) ✗ pio upgrade
You're up-to-date!
PlatformIO 3.6.0 is currently the newest version available.
➜  gate-weemos git:(master) ✗ pio run
Processing gate-wemos (platform: atmelavr; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------
Error: Unknown board ID 'd1_mini'
➜  gate-weemos git:(master) ✗ bat platformio.ini
───────┬──────────────────────────────────────────────────────────────────────────
       │ File: platformio.ini
───────┼──────────────────────────────────────────────────────────────────────────
   1   │ [env:gate-wemos]
   2   │ platform = atmelavr
   3   │ framework = arduino
   4   │ board = d1_mini
   5   │ lib_deps =
   6   │   PubSubClient@2.6
   7   │   MFRC522@1.3.6
───────┴──────────────────────────────────────────────────────────────────────────
```

bendem, 2018-09-29:
nevermind, I used the wrong platform, setting `platform = espressif8266` worked