platformio/platformio-core #1445: unable to build solution after VSCODE IDE Update

enny1170, 2018-03-13:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [x ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: W10 x64

**PlatformIO Version** (`platformio --version`): 3.5.2rc4

### Description of problem
After updating the IDE Plugin try to Build runs in fallowing Error:
> Executing task: C:\Python27\Scripts\platformio.exe run <

Traceback (most recent call last):
  File "C:\Python27\Scripts\platformio-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3195, in <module>
    @_call_aside
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3179, in _call_aside
    f(*args, **kwargs)
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3208, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 683, in _build_master
    return cls._build_from_requirements(__requires__)
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 696, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 885, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==3.5.2rc2' distribution was not found and is required by the application
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

#### Steps to Reproduce

1. Intalls VSCO PlattformIo IDE Plugin
2. Build a Project
3.

### Actual Results
> Executing task: C:\Python27\Scripts\platformio.exe run <

Traceback (most recent call last):
  File "C:\Python27\Scripts\platformio-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3195, in <module>
    @_call_aside
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3179, in _call_aside
    f(*args, **kwargs)
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 3208, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 683, in _build_master
    return cls._build_from_requirements(__requires__)
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 696, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "c:\users\e.ziehe\.platformio\penv\lib\site-packages\pkg_resources\__init__.py", line 885, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==3.5.2rc2' distribution was not found and is required by the application
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

ivankravets, 2018-03-13:
Please open PIO IDE Terminal and type `pio --version`. Does it work?

enny1170, 2018-03-13:
Returns v 3.5.2rc4. See below.

C:\Users\e.ziehe\Documents\arduino >pio --version
PlatformIO, version 3.5.2rc4

ivankravets, 2018-03-13:
I know the solution for you. I've just want to reproduce issue. Please try now:
```
pio -c vscode -f --version
```

enny1170, 2018-03-13:
That helps a bit. Now i can run build and upload from terminal window.
But the Problem will be the same if i use the platformio:Build commad/Task inside vscode.

Do you know also a solution ?

Regards Enrico

ivankravets, 2018-03-13:
Could you provide https://www.teamviewer.com/en/ session (ID/PAss) I'll try to reproduce this issue and find a bug?

Please mail me@ikravets.com. I would be thankful!

ivankravets, 2018-03-13:
The problem was linked with multiple PIO Cores http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

closed 2018-03-13 by ivankravets