platformio/platformio-core #1447: Local path for packages

kylefleming, 2018-03-13:
What kind of issue is this?

- [x] PlatformIO Core.

------------------------------------------------------------------

### Configuration

**Operating system**:

MacOS 10.13.3

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 3.5.2rc4

### Description of problem

I would like to be able to specify a local path for packages. Currently, it seems the way to do that is to have a local copy of the platform and specify the path for that platform in the platformio.ini. Then, inside the platform's platform.json, set the version of the package to the path for that package (prepended with `file://`).

However, when I do that, it copies the folder of the package to the `~/.platformio/packages` folder, preventing me from making further changes to my package's original package location, without deleting the copied version between every build or editing the version in the `~/.platformio/packages` folder.

Ideally, I would like to be able to keep the original location of the package that I specified and keep editing from that location.

ivankravets, 2018-03-14:
Do you mean https://github.com/platformio/platformio-core/issues/1367 ?

kylefleming, 2018-03-14:
Ah yes, I think that will do it. Thank you

closed 2018-03-14 by kylefleming