platformio/platformio-core #1454: PIO Home board selection in project wizard and sort of project folders

sstaub, 2018-03-16:
Home v0.8.1
Core v3.5.3a3

The search function for boards in the project wizard should ignore lower/upper cases and spaces, e.g.:
"mbed lpc1768" gives no result, also "mbed LPC 1768", only "mbedLPC1768" gives the right result.

The sort function of the folders and data in the location depends on lower/upper cases, it confusing because the os (macOS) sorts without depends of cases.
[image]

ivankravets, 2018-03-16:
> only "mbedLPC1768" gives the right result.

Are you sure? It does not work for me. The search is case-insensitive.

[image]

sstaub, 2018-03-16:
Interesting, now it works. Maybe there is an complete restart necessary for VSCode after an update.

ivankravets, 2018-03-22:
Please open PIO IDE Terminal and type
```
pio upgrade --dev
pio update
```

Restart IDE.

commit ca34da51 (2018-03-22): PIO Home: sort folders by name (ignore case), catch ServerError exceptions // Resolve #1454 Resolve #1302
closed 2018-03-22 by ivankravets