platformio/platformio-core #1459: Home: Internal Store Exception

mariusvw, 2018-03-19:
TypeError: Cannot read property 'name' of null
    at http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:47094
    at Array.map (native)
    at b (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:46965)
    at http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:47484
    at Array.map (native)
    at _ (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:47424)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:17:244163)
    at r (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:10153)
    at l (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:11523)
    at u (http://127.0.0.1:8010/0fce13c15ea1c9f96851.min.js:6:11641)

ivankravets, 2018-03-20:
Could you reproduce this issue again?

mariusvw, 2018-03-22:
Solved it by clearing out the compile-cache. Can't reproduce it since then.

closed 2018-03-22 by mariusvw