platformio/platformio-core #1464: Home: Could not load PIO Account information

RayEbb, 2018-03-19:
Server error. I did a password reset, but i can't login with the temp Password.

ivankravets, 2018-03-20:
Please open PIO IDE Terminal and run `pio upgrade`. Restart IDE. Does it work now?

RayEbb, 2018-03-20:
Thank you very much, IvanKravets!! I tried to start the PIO terminal, but the only terminal i saw, was the CMD.  That's why i uninstalled de PIO IDE, and reinstalled it again. I think i have to setup Visual Studio Code first, and the Workbench also. But it's very tricky for me, to be honest. So many settings, lol... And now, i found the little PIO icons on the bottom of the screen, with the Terminal at the end of the line.... 

And now, IT WORKS!!!! 
Stupid me, i know.. :D But, I've learned something, thanks to you!!!

ivankravets, 2018-03-20:
Happy coding with PlatformIO!

closed 2018-03-20 by ivankravets

RayEbb, 2018-03-20:
Thx!! But there's a very long road to go. :D