platformio/platformio-core #1474: Home: Could not load latest Tweets

epsimatic, 2018-03-21:
I've blocked your app access to twitter.com and this pops up every time I open «Home» page. I doupt it's so important message to show it blocking a big part of the main view

HTTPSConnectionPool(host='twitter.com', port=443): Max retries exceeded with url: /i/profiles/show/PlatformIO_Org/timeline/tweets?include_available_features=1&include_entities=1&include_new_items_bar=true (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x10c54d6d0>: Failed to establish a new connection: [Errno 64] Host is down',))

ivankravets, 2018-03-21:
Sorry, we will fix it ASAP.

ivankravets, 2018-03-22:
Please open PIO IDE Terminal and type `pio update`. Restart IDE.

commit f937eabc (2018-03-22): Handle error connections to the latest news for PIO Home // Resolve #1470 Resolve #1474 Resolve #1478 Resolve #1480
closed 2018-03-22 by ivankravets