platformio/platformio-core #1516: vscode (1.21.1)+platformio  (3.5.3a7) + openocd (0.10) -> no jlink device found

ozanoner, 2018-03-23:
What kind of issue is this?

- PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: windows 10

**PlatformIO Version** (`platformio --version`): 3.5.3a7

### Description of problem
unable to upload firmware.hex

#### Steps to Reproduce

1. create a new project for nrf52-dk with vscode+platformio
2. try example code: .platformio\platforms\nordicnrf52\examples\mbed-events\src
3. pio run target --upload

### Actual Results
   
[03/23/18 22:51:47] Processing nrf52_dk (platform: https://github.com/platformio/platform-nordicnrf52.git; board: nrf52_dk; framework: mbed)

Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Nordic nRF52 > Nordic nRF52-DK
SYSTEM: NRF52832 64MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(jlink) ON-BOARD(cmsis-dap, jlink) EXTERNAL(blackmagic, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mbed-events>
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, nrfjprog, stlink
CURRENT: upload_protocol = jlink
Uploading .pioenvs\nrf52_dk\firmware.hex
GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-10:20)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
0
adapter speed: 1000 kHz
cortex_m reset_config sysresetreq
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Warn : Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
*** [upload] Error -1
================================================================= [ERROR] Took 5.63 seconds =================================================================
The terminal process terminated with exit code: 1

### Expected Results
uploaded firmware

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nrf52_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = mbed
board = nrf52_dk
debug_tool = jlink
upload_protocol = jlink
```

**Source file to reproduce issue:**
```cpp
#include "mbed_events.h"
#include <stdio.h>

int main() {
    // creates a queue with the default size
    EventQueue queue;

    // events are simple callbacks
    queue.call(printf, "called immediately\n");
    queue.call_in(2000, printf, "called in 2 seconds\n");
    queue.call_every(1000, printf, "called every 1 seconds\n");

    // events are executed by the dispatch method
    queue.dispatch();
}

```

### Additional info
[image]

ivankravets, 2018-03-23:
Do you see J-LInk in system device manager?

ozanoner, 2018-03-23:
yes
[image]

ozanoner, 2018-03-23:
I installed drivers here: C:\Program Files (x86)\SEGGER\JLink_V630h\USBDriver

ivankravets, 2018-03-23:
https://sourceforge.net/p/openocd/mailman/message/32252376/

ivankravets, 2018-03-23:
http://zadig.akeo.ie/

ozanoner, 2018-03-23:
yeah. actually I tried it, and worked but with a side-effect (and a terrible one). it causes a disabled com port somehow, which means there is no terminal access to the nrf52-dk. rollback took my 3 hours, therefore, I won't try it once more :)
there is a serious warning here: 
[image]
https://gnu-mcu-eclipse.github.io/openocd/install/

which I didn't take serious first (causing 3 wasted hours)

ivankravets, 2018-03-23:
Indeed, you can use J-LInk GDB server directly, without openOCD. Could you try it?

* http://docs.platformio.org/en/latest/plus/debugging.html#j-link-and-st-nucleo

ozanoner, 2018-03-23:
ok let me try. do I need "extra_script.py"? if so, does it need any edit?

ivankravets, 2018-03-23:
No no, just use `debug_server`. You even can keep `debu_tool = stlink`.

ozanoner, 2018-03-23:
seems that I'm doing something wrong. can you please check the following ini?

[image]

ivankravets, 2018-03-23:
1. `debug_tools = jlink`
2. Please remove quotes from debug server arguments

ozanoner, 2018-03-23:
I cannot get rid of openocd :(

latest ini:

[env:nrf52_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = mbed
board = nrf52_dk
; debug_tool = jlink
upload_protocol = jlink

debug_tool = jlink
debug_server =
  C:/Program Files (x86)/SEGGER/JLink_V630h/JLinkGDBServerCL.exe
  -singlerun
  -if
  SWD
  -select
  USB
  -port
  2331
  -device
  nRF52832_xxAA

and terminal output:

Uploading .pioenvs\nrf52_dk\firmware.hex
GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-10:20)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
0
adapter speed: 1000 kHz
cortex_m reset_config sysresetreq
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Warn : Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
*** [upload] Error -1

actually segger debugger works from command line:
[image]

if I am not wrong, "debug_server" option doesn't change the upload utility (which is openocd now)

ivankravets, 2018-03-23:
This works for me:
```ini
[env:nrf52_dk]
platform = nordicnrf52
framework = mbed
board = nrf52_dk
debug_server =
    C:/Program Files (x86)/SEGGER/JLink_V630h/JLinkGDBServerCL.exe
    -singlerun
    -if
    SWD
    -select
    USB
    -port
    2331
    -device
    nRF52832_xxAA
```

ozanoner, 2018-03-23:
can you post terminal output too?

ivankravets, 2018-03-23:
[image]

ozanoner, 2018-03-23:
seems that it doesn't work for you either. can you show me 'TERMINAL' tab pls?

[image]

ozanoner, 2018-03-23:
I really appreciate if you can upload the firmware under windows-10 with a proper ini file. 
I've tested something new. jlink.exe works, but I couldn't configure platformio to use jlink.exe. 
[image]

do you know how to configure platformio to use jlink as 'uploader'. I tried this ini:
```
[env:nrf52_dk]
platform = https://github.com/platformio/platform-nordicnrf52.git
framework = mbed
board = nrf52_dk

upload_protocol = custom
extra_scripts = extra_script.py

debug_tool = jlink

debug_server =
  C:/Program Files (x86)/SEGGER/JLink_V630h/JLinkGDBServerCL.exe
```
and this script:
```
Import("env")

env.Replace(
    UPLOADHEXCMD=
    'JLink.Exe -device nRF52832_xxAA -speed 4000 -if swd -autoconnect 1 -CommanderScript $BUILD_DIR/1.jlink'
)
```
but no luck :( still it uses openocd and fails:
Uploading .pioenvs\nrf52_dk\firmware.hex
GNU MCU Eclipse 32-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-10:20)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
0
adapter speed: 1000 kHz
cortex_m reset_config sysresetreq
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Warn : Failed to open device: LIBUSB_ERROR_NOT_FOUND.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
*** [upload] Error -1

is there any way to select & configure uploader tool?
I see this in the platform.json: 
https://github.com/platformio/platform-nordicnrf52/blob/c45f941e1b076272ceecbdd2d6fd4b575ac5517c/platform.json#L48

and this:
https://github.com/platformio/platform-nordicnrf52/blob/c45f941e1b076272ceecbdd2d6fd4b575ac5517c/platform.json#L53

how can I config to use 'tool-nrfjprog' as uploader? I think this may solve my problem.

ozanoner, 2018-03-23:
just to clarify: my problem is not with debugger, but uploader. openocd as uploader fails in my trials and I am trying to replace openocd with segger tools to be able to upload in platformio (with ctrl+alt+U)

ivankravets, 2018-03-23:
Oh... Upload!!! I thought you mean debugging! Does debug work for you?
Of course uploading will not work. You need to declare custom uploader. For example,

```
[env:]
..
upload_protocol = custom
extra_scripts = extra.py
```

`extra.py`
```py
from os import makedirs
from os.path import isdir, join
Import(env)

def _jlink_cmd_script(env, source):
    build_dir = env.subst("$BUILD_DIR")
    if not isdir(build_dir):
        makedirs(build_dir)
    script_path = join(build_dir, "upload.jlink")
    commands = ["h", "loadbin %s,0x0" % source, "r", "q"]
    with open(script_path, "w") as fp:
        fp.write("\n".join(commands))
    return script_path

env.Replace(
    __jlink_cmd_script=_jlink_cmd_script,
    UPLOADER="C:/Program Files (x86)/SEGGER/JLink_V630h/JLinkGDBServerCL.exe",
    UPLOADERFLAGS=[
        "-device", "nRF52832_xxAA",
        "-speed", "4000",
        "-if", "swd",
        "-autoconnect", "1"
    ],
    UPLOADCMD='"$UPLOADER" $UPLOADERFLAGS -CommanderScript ${__jlink_cmd_script(__env__, SOURCE)}'
)
```

ivankravets, 2018-03-23:
P.S: I didn't test it :) Just wrote.

ozanoner, 2018-03-23:
thumbs up!! many thanks @ivankravets . your solution works with small modifications. I am attaching working platform.ini and extra_script.py for others.

[tmp.zip](https://github.com/platformio/platformio-core/files/1843677/tmp.zip)

ivankravets, 2018-03-23:
Stop! Maybe, we should use official JLink GDB default without these extra scripts and hooks? PIO should work smoothly.

ozanoner, 2018-03-23:
ok, right. I've used jlink.exe as uploader, not jlink gdb. moreover, I didn't test jlink-gdb as debugger. how can I help you?

profra, 2018-03-24:
Hi all, please do it according to Ivan ... "Stop! Maybe, we should use official JLink GDB default without these extra scripts and hooks? PIO should work smoothly." ... I have been solving a similar problem right now... the clear solution would be beneficial to all!

ozanoner, 2018-03-24:
sure, he is the boss :)

profra, 2018-03-24:
MVP for PIO! 🥇 👍

ivankravets, 2018-03-24:
Please remove all extra scripts and extra lines from `platformio.ini` and re-test with upstream version. Don't forget to add JLink installation to your system PATH. `JLinkExe` command should from your `cmd` terminal without the full path.

* http://docs.platformio.org/en/latest/platforms/nordicnrf52.html#stable-and-upstream-versions

ivankravets, 2018-03-24:
If you already use upstream version, please PIO Home > Platforms > Updates.

All should work out the box now.

profra, 2018-03-24:
I am in weekend-house now... My equipment here is only NB with Win10 Pro and Nucleo-64 STM32F103 with ST/link on board... and VS Code with PIO... I would like to rid off OpenOCD (now is working OK) on this board, I would prefer to have direct GDB as uploader and debugger. Do you see any possibilities how to reach it? ... and how?

ivankravets, 2018-03-24:
>  I would prefer to have direct GDB as uploader and debugger.

```
pio run -t upload
pio debug --interface gdb -x .pioinit
```

ivankravets, 2018-03-24:
Please reopen if you still need a help

closed 2018-03-24 by ivankravets

ozanoner, 2018-03-24:
@ivankravets  sorry man, my bad about jlink.
the right tool should be nrfjprog, not jlink.
The following command worked correctly.
nrfjprog --program firmware.hex --chiperase --reset

Of course, you can update the repo for that tool, but if you wait, I will fully test it within script and confirm if it works as so.  
[image]

ivankravets, 2018-03-24:
`upload_protocol = nrfjprog`

profra, 2018-03-24:
I would prefer to have direct GDB as uploader and debugger.

pio run -t upload
pio debug --interface gdb -x .pioinit
---------------------------------------------------
OK, but I don't know how to connect (gdb) console... any help?

ivankravets, 2018-03-26:
You have! We use GDB for debugging.