platformio/platformio-core #1532: Does lib_extra_dirs  support relative path?

palivoda, 2018-03-26:
I have project structure where 

src/main.c
lib/foo/src/a.h
lib/foo/lib/woo/src/b.h

main.c includes a.h
a.h includes b.h

plaformio.ini:

[platformio]
lib_extra_dirs = lib/woo/lib

And I get error b.h is not found. 

Does lib_extra_dirs should work in such case?
How I can include b.h?

ivankravets, 2018-04-07:
`lib` folder MUST contain ONE library per folder. You can use nested folders here

closed 2018-04-07 by ivankravets

palivoda, 2018-04-16:
> You can use nested folders here

Hi Ivan,  can you clarify a bit?

From response I assume I even dont need lib_extra_dirs  option and I can just put

/lib/FooLib
and inside FooLib there could be also lib folder
/lib/FooLib/lib/ZooLib/src/aaa.h

this aaa.h will be included?

ivankravets, 2018-04-17:
You should have 1 level libraries structure. 

- lib/ZooLib/src/aaa.h
- lib/BooLib/src/bbb.h

palivoda, 2018-04-17:
Then it's impossible to include other project as git submodule.

ivankravets, 2018-04-20:
You can use http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-extra-dirs as you described above. To eliminate whole library(with sublibraries) building, just ignore it via http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

nelfata, 2018-08-25:
On a similar issue, if I have a precompiled library and I am using the following:

-L../../Libraries -llibSomeLib.a

the relative path does not work, I have to specify the absolute path in the -L directive.

Is this a bug, or is there another method to link with a precompiled library?

ivankravets, 2018-08-31:
@nelfata I've just tried and can't reproduce this issue. Please provide a test example and open a new issue.