platformio/platformio-core #1533: C/C++ project index rebuild failed

pleink, 2018-03-27:
Error: Traceback (most recent call last):
  File "c:\users\tizi\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\tizi\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\users\tizi\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 120, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\users\tizi\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 146, in _merge_contents
    with open(dst_path, "w") as f:
IOError: [Errno 13] Permission denied: 'G:\\Arduino\\Top Projects\\GreenPlant 0.4\\.clang_complete'

I have two computers and always when I import a Projekt from the other computer I get a project index rebuild fail

ivankravets, 2018-04-07:
It seems like a permission issue. Please remove `G:\Arduino\Top Projects\GreenPlant 0.4\.clang_complete` manually.

closed 2018-04-07 by ivankravets

pleink, 2018-04-09:
thanks for the answer, I found the mistake on my own. I don't know how that
could happen, but sometimes when I move a Project only the .pioenvs, lib,
src and platformio folder/ config file gets moved.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>