platformio/platformio-core #1535: the package manager does not respect OS release version

trombik, 2018-03-28:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

N/A

**Operating system**:

FreeBSD (also applies to OpenBSD).

**PlatformIO Version** (`platformio --version`):

```
PlatformIO, version 3.5.1
```

### Description of problem

platform.io does not respect ABIs of OS. FreeBSD keeps ABI compatibility in minor releases. a package for release `N.M` works on release `N.M+1`, but not necessarily for `N+1.0`. it might work sometimes, but not always. the package manager only respects OS name and `${ARCH}` (see https://github.com/platformio/platformio-core/blob/ec9a2b02eac01a41098b6de4cc01d1de17b26c59/platformio/util.py#L176). the `system` attribute in `package.json` should include major release version so that different release version installs packages built for the major release, something like `FreeBSD_11_amd64`.

#### Steps to Reproduce

1. install a platform.io package on a FreeBSD release, preferably CURRENT since most of, if not all, packages were built on a release.
2. see the release version the package was built for, such as `file ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc`. the file come with `toolchain-xtensa` version `1.40802.0`
3. it shows unmatched release version

### Actual Results

`uname -r` on my machine says `12.0-CURRENT`, but `file(1)` says the binary was built for FreeBSD 9.2:

```
ELF 64-bit LSB executable, x86-64, version 1 (FreeBSD), dynamically linked, interpreter /libexec/ld-elf.so.1, for FreeBSD 9.2 (902502), stripped
```

in this case, the binary _happened to run_ fine on my machine, probably on 10.x and 11.x. but when ABI changes, it will break.

this issue affects OpenBSD hard. in OpenBSD, ABI compatibility is not kept even in minor releases, and they are known to be aggressive to bump library versions in the base system. for OpenBSD, minor release version should be included.

### Expected Results

the version in the output should match the major release version of the OS.

an example of expected results:

```
ELF 64-bit LSB executable, x86-64, version 1 (FreeBSD), statically linked, for FreeBSD 12.0 (1200060), FreeBSD-style, stripped
```

### If problems with PlatformIO Build System:

N/A

### Additional info

the expected result above is one from [`toolchain-xtensa32`](https://github.com/trombik/toolchain-xtensa32) that I am currently building for FreeBSD, which is not available in the platform.io package repository. note that package for FreeBSD CURRENT may be missing in the official package repository for obvious reasons. the point is, packages are built for a particular release.

CyanoFresh, 2020-01-10:
Bump
Also need toolchain-xtensa32 for esp32 on freebsd

myfreeweb, 2020-06-24:
FreeBSD keeps backwards compatibility for ages, you can run even 4.0 binaries on a -CURRENT kernel just fine, provided you have all the required libraries (from that version or backwards compatible ones).

The builds on platformio are bad though: since they were built on 9.x, they are dynamically linked to `libstdc++` (FreeBSD switched to `libc++` ages ago.. in 10 to be precise). And what we get is:

```
ld-elf.so.1: Shared object "libstdc++.so.6" not found, required by "xtensa-lx106-elf-g++"
```

A workaround is installing GCC, which includes `libstdc++`, which is still backwards compatible (SOVERSION `.6`). And adding it to `LD_LIBRARY_PATH`:

```
% LD_LIBRARY_PATH=/usr/local/lib/gcc9 ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc
xtensa-lx106-elf-gcc: fatal error: no input files
compilation terminated.
```

With that, I've been able to use ESPHome to compile ESP8266 firmware. It's actually pretty cool that this platformio thing supports FreeBSD, but the binaries really shouldn't be built on `<10` anymore.

trombik, 2020-06-24:
platformio's binaries have more issues which affect not only FreeBSD, but also others. see https://github.com/trombik/platformio-freebsd-toolchain-xtensa#rationale