platformio/platformio-core #1542: [Question] How can I flash firmware.bin manually by using `esptool.py` after build by `platformio run --target build`?

dphans, 2018-03-30:
### Configuration

**Operating system**: MacOS 10.13.3

**PlatformIO Version**: 3.5.2

### Description of problem
After build project by running `platformio run --target build`. I found the `firmware.bin` file created at `.../PlatformIO/Projects/<my-project>/.pioenvs/<board-name>/firmware.bin`.
How can I upload `firmware.bin` manually by running `esptool.py` command from external Terminal program?
Because I'm going to write a tool can flash `firmware.bin` for many users simply.
I know that not only just upload `firmware.bin` file, it must includes `partitions.bin` and `bootloader.bin`. But I don't know correct address and offset for each binary files.

This is not issues, just my problem. I'm sorry if this post is wrong.

ivankravets, 2018-04-07:
Please move to https://community.platformio.org/

closed 2018-04-07 by ivankravets

probonopd, 2021-04-13:
Running into the same issue.

Is it possible to combine all parts into  __one__ firmware image that contains __everything__ and can be easily be flashed to the ESP32 using `esptool.py` starting at offset `0x00`? Not a bunch of files in different directories.

ivankravets, 2021-04-13:
See [upload_command](https://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-command)

45gfg9, 2022-11-07:
Same problem here. I want to create a .bin file, that doesn't just contain the app code section, but also the filesystem image, nvs, bootloader etc that just make up the whole 4MB (or larger) flash chip size, that I can just flash it into the address 0x0 of the ESP32. Is that possible or am I having some misconception here?

gsicilia82, 2023-01-08:
This would be a great feature. I encounter same issue. I want to provide firmware which can be flashed as simple as possible.

ivankravets, 2023-01-09:
You can use recently added [pio pkg exec](https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_exec.html) command

```
pio pkg exec --package "platformio/tool-esptoolpy" -- esptool.py --help
```