platformio/platformio-core #1546: pio generates wrong __cplusplus value in c_cpp_properties.json resulting in IDE build errors for std::unique_ptr in Visual Studio Code

jeraymond, 2018-04-01:
### Configuration

**Operating system**: macOS High Sierra

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.3a7, PlatformIO IDE v0.14.0

### Description of problem

When opening projects in PlatformIO IDE in Visual Studio Code, the generated c_cpp_properties.json is set with value `"__cplusplus=199711L"`. This causes C++11 features such as std::unique_ptr not to be recognized by the IDE. Using unique_ptr shows as build error in the IDE, but the code builds fine via the command line (pio run).

Manually updating the value to `“__cplusplus=201103L”` makes the errors in the IDE go away.

#### Steps to Reproduce

1. In Visual Studio Code import project: https://github.com/jeraymond/problem-project
2. Perform a build in the IDE.

### Actual Results

Build error in IDE: 

`"std" has no member "unique_ptr"'`

### Expected Results

Complication success.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:huzzah]
build_flags = -Wl,-Teagle.flash.4m.ld
platform = espressif8266
board = huzzah
framework = arduino
monitor_baud=115200
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <memory>

void setup() {

}

void loop() {
  size_t size = 1024;
  // Allocate a buffer to store contents of the file.
  std::unique_ptr<char[]> buf(new char[size]);
}
```

### Additional info

There is a community thread about this issue https://community.platformio.org/t/error-in-vscode-but-not-in-command-line-build/3803/6.

Thanks to thaumiel for providing a workaround for the issue. Manual change `"__cplusplus=199711L"` to `"__cplusplus=201103L"` in `c_cpp_properties.json` and the IDE compile errors go away.

jeraymond, 2018-04-01:
Maybe the value should really be `"__cplusplus=201703L"`. Elsewhere in `c_cpp_properties.json` I see `"cppStandard": "c++17"`. These version must match.

jeraymond, 2018-04-01:
Reopening the project causes ‘c_cpp_properties.json’ to be overridden reverting manual workaround modifications to the file.

ivankravets, 2018-04-01:
Please run in PIO IDE Terminal: `pio upgrade --dev`. 

P.S: I propose to keep this issue as opened and implement support for 0.16 .0 version of https://github.com/Microsoft/vscode-cpptools

jeraymond, 2018-04-02:
Thanks for the quick fix. How soon will this make it into a release?

ivankravets, 2018-04-02:
@jeraymond 3.5.2 release is not affected. This change was introduced a few days ago.

ivankravets, 2018-04-20:
Resolved in https://github.com/platformio/platformio-core/commit/b77160d36342e438708050a0fbff52a13b9b4a2c

ivankravets, 2018-04-20:
Please open VSCode > Tasks > PlatformIO: Upgrade PIO Core. Retsart VSCode.

closed 2018-04-20 by ivankravets