platformio/platformio-core #1557: After updating VSCode the build task is not working anymore

fmuntean, 2018-04-09:
### Configuration

**Operating system**: 
Windows 10 x64

**PlatformIO Version** (`platformio --version`):
- [ ] PlatformIO IDE. 
Home 0.9.3
Core 3.5.3a8

**VS CODE Version**
Version 1.22.1
Commit 950b8b0d37a9b7061b6f0d291837ccc4015f5ecd
Date 2018-04-06T00:27:19.487Z
Shell 1.7.12
Renderer 58.0.3029.110
Node 7.9.0
Architecture ia32
 

### Description of problem
The build task that was working few minutes before is stuck with no progress after upgrading the VSCode to Version 1.22.1

I am using both the ESP8266 and MXCHIP AZ3166 and neither projects are building anymore.

#### Steps to Reproduce

1. upgrade the VS code to latest version
2. restart VSCODE
3. open a project and build 

### Actual Results
The executing task gets stuck with no progress for one hour now.

### Expected Results
Should complete the build or give an error with a reason.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
[env:mxchip_az3166]
platform = ststm32
board = mxchip_az3166
framework = arduino

**Source file to reproduce issue:**

Any source file 

### Additional info
This is the only message appearing in the Task - Build window
> Executing task: D:\GIT\HomeAutomation\AZ3166\Remote_Monitoring\platformio.exe run <

gmdriscoll, 2018-04-09:
I am having the same problem.  

Further information: 
Build, Upload, Serial Monitor do not give any info in tasks window.  
Unable to end build tasks when clicking checkmark
Able to run build and upload manually from command prompt with "pio run", etc.
Tried using CTRL-ALT-U  or  CTRL-ALT-B and get this message:

Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "args": [
        "run",
        "--target",
        "upload"
    ],
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.

AlexYukikaze, 2018-04-09:
Same problem
```> Executing task: C:\esp\WiFiRelay\platformio.exe run --target upload <```

ivankravets, 2018-04-12:
Please upgrade PlatformIO IDE VSCode extension to 0.14.1

closed 2018-04-12 by ivankravets

wernersmook, 2019-04-21:
The upload is working, but I get this message.  Can someone tell me perhaps how to fix it?

[image]

OUTPUT:

Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ],
    "group": {
        "kind": "build",
        "isDefault": true
    }
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Upload",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Clean",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Monitor",
    "problemMatcher": [
        "$platformio"
    ],
    "group": {
        "kind": "build",
        "isDefault": true
    }
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Upload",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Clean",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.

ivankravets, 2019-04-22:
@wernersmook 
1. Please check that you use the latest VSCode and PlatformIO IDE extension
2. Check that you opened PlatformIO based project which contains `platformio.ini` in the root.

wernersmook, 2019-04-22:
Good day Ivan

1 I have installed VSC 1.33.1 and PIO 1.7.1, according to me the latest of both

After my previous post I encountered a problem with the homescreen that doesn't load.  After a (long) while (several restarts) I got an error screen containing the message beneath.  I got fed up and decided to uninstall/reinstall PIO extention.  Things seems to be ok for now and I will try to do some things on it today.

Thanks for reply.

++++++++++++++++++++++++++++++++++++++++++

C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\server.py:199 in process
198                    self._encoder = encoder
199            self.render(resrc)
200        except:
C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\server.py:259 in render
258        try:
259            body = resrc.render(self)
260        except UnsupportedMethod as e:
C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\resource.py:250 in render
249            raise UnsupportedMethod(allowedMethods)
250        return m(request)
251
<string>:448 in render_GET
447
448
449
C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\http.py:1271 in setHeader
1270        """
1271        self.responseHeaders.setRawHeaders(name, [value])
1272
C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:220 in setRawHeaders
219        encodedValues = [_sanitizeLinearWhitespace(v)
220                         for v in self._encodeValues(values)]
221
C:\Users\werne\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:40 in _sanitizeLinearWhitespace
39    """
40    return b' '.join(headerComponent.splitlines())
41

ivankravets, 2019-04-22:
Just run PlatformIO IDE > Activity Bar (left sidebar) > PlatformIO > Updates > Update all

mklengel, 2019-05-02:
Hi Ivan,

same error on my system (macOS 10.14.4, VS Code 1.33.1, PlatformIO IDE 1.7.1):

```
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Upload",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
```

Contents of platformio.ini:
```
[env:uno]
platform = atmelavr
board = uno
framework = Arduino
```

Build:
```
Executing task in folder mkBlinkyMega2560: platformio run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz 8KB RAM (248KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 5 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/megaatmega2560/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.1% (used 9 bytes from 8192 bytes)
PROGRAM: [          ]   0.6% (used 1568 bytes from 253952 bytes)
======================================================================================== [SUCCESS] Took 0.76 seconds ========================================================================================

Terminal will be reused by tasks, press any key to close it
```

DorianGrey2012, 2019-08-27:
> @wernersmook
> 
> 1. Please check that you use the latest VSCode and PlatformIO IDE extension
> 2. Check that you opened PlatformIO based project which contains `platformio.ini` in the root.

I have the latest update of Platformio core ad VScode, I have platformio.ini in the root of the project.

I havethe same error as @wernersmook has/had.

ivankravets, 2019-08-29:
Please remove `.vscode` folder from project and restart VSCode.

myLens, 2021-09-23:
I used to be quite happy with VS-code / platformio but what is happening now tells me to better rely on paid solutions one can trust not only today but also tomorrow. 
All the tips and hints given to sort out the sudden incompatibility did not work yet and the time spent because a developer forgot to include proper migration tools is more expensive than a clean and professional solution.
This stuff is only good for hobbyists !

ivankravets, 2021-09-23:
@myLens sorry for the issue. Could more details on what could be improved?