platformio/platformio-core #1564: SRC_FILTER not workin for arduino project

zafrirron, 2018-04-15:
reproduction

1. Create arduino project.
2. Define src_dir = source
3. put some file.ino  in source/subfolder
4. Define (env level) SRC_FILTER = +<subfolder/> (or +<subfolder/*> or +<subfolder/*.*> or +<subfolder/*.ino> or +<subfolder/*.ino.cpp> or +<subfolder/file.ino>

The source files are not collected...
changing file.ino names to file.cpp does include them but not in arduino model (not same as putting file.ino into src_dir)

ivankravets, 2018-04-17:
INO is a fake format intended for Arduino IDE. This is not a valid CPP file. PlatformIO converts this file to CPP. As result, you need to operate with CPP files instead of INO... `<subfolder/.ino.cpp>`

Also, I highly recommend converting all files to CPP where you will receive true code completion and code linting services http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

closed 2018-04-17 by ivankravets

zafrirron, 2018-04-17:
Thanks,
Agree with all your points, having said that two points to consider:
1. All arduino IDE developers should move from that stupid environment to (preferably platformio ),
in order to do this the migration path should be as simple as possible:
  a. import project to platformio (ino files converted (by InoToCPPConverter in piomisc.py).
  b. for large arduino project, move to more reasonable folder structure.
  c. start to migrate to cpp
2. once someone moves out from that S ide, there shouldn't be any way back, step b is not back, this to prevent back and forth between the PIO IDE and arduino IDE....