platformio/platformio-core #1573: [ ] PlatformIO Core

JeffreyStoel, 2018-04-16:
- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.
------------------------------------------------------------------
### Configuration

**Operating system**: 
Linux
**PlatformIO Version** (`platformio --version`):
No version
### Description of problem
Whenever I try to install PIO with Trizen I get this weird error message:
  File "<string>", line 93
    print r['out']

I'm not a very experienced programmer or user so I don't really know what to do, honestly
I don't think I can even try to solve the program, but maybe someone else encountered this
before and knows how to fix this bug/error. I tried installing different kinds of python, but that didn't seem to solve the problem.

Thanks in advance.

JeffreyStoel, 2018-04-17:
@ivankravets 

[env:mini]
platform = atmelavr
board = pro16MHzatmega328
upload_port = /dev/ttyUSB0
framework = arduino
targets = upload

[env:nano]
platform = atmelavr
board = nanoatmega328
upload_port = /dev/ttyUSB0
framework = arduino
targets = upload

[env:uno]
platform = atmelavr
board = uno
upload_port = /dev/ttyUSB0
framework = arduino
targets = upload

[env:mega]
platform = atmelavr
board = megaatmega2560
upload_port = /dev/ttyACM0
framework = arduino
targets = upload

[env:esp]
platform = espressif8266
board = nodemcuv2
framework = arduino
upload_port = /dev/ttyUSB0
targets = upload

My OS is Arch Linux

ivankravets, 2018-04-17:
It seems that your package manager installed PIO incorrect. It should use Python 2.7 instead of Python. Please contact package maintainer or install PIO Core using `pip2 install platformio` command.

closed 2018-04-17 by ivankravets