platformio/platformio-core #1577: [Feature Request] Make it easy to modify existing platformio.ini variables

ssilverman, 2018-04-17:
**Operating system**:
10.13.4

**PlatformIO Version** (`platformio --version`):
3.5.3a9

### Description of problem
This started with a need to exclude one file from a build. Writing it as:
`src_filter = -<filetoexclude.cpp>`
means that `+<*>` won't get included, so the project doesn't build. This means that what I really want is to append `-<filetoexclude.cpp>` to the existing `src_filter`.

I tried doing a few things similar to:
`src_filter = ${src_filter} -<filetoexclude.cpp>`
and:
```
[common]
src_filter = ${src_filter}

[env:myproject]
src_filter = ${common.src_filter} -<filetoexclude.cpp>
```
but those don't work.

So then I thought that if I just want to append to an existing default, it would be nice if there were some syntax to do that. For example:
`src_filter = ${defaults.src_filter} -<filetoexclude.cpp>`

If this doesn't sound like a good idea, is there a more convenient way to exclude files from a build without having to rewrite what was already in the default value? I dislike copying & pasting because the defaults may change, hence this feature request.

ivankravets, 2018-04-17:
http://docs.platformio.org/en/latest/projectconf/dynamic_variables.html

```ini
[common]
src_filter = +<*>

[env:myproject]
src_filter = ${common.src_filter} -<filetoexclude.cpp>
```

closed 2018-04-17 by ivankravets

ssilverman, 2018-04-17:
Yes, that works, but it requires having to rewrite src_filter. I was suggesting a solution where it doesn’t have to be rewritten.