platformio/platformio-core #1598: PlatformIO: Update platforms, libs and packages doesn’t work after VSCode update

DJManas, 2018-04-28:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS 10.13.4

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.3b1

```
### Description of problem
When in VSCode I give cmd+shift+p and select PlatformIO: **Platformio: Update installed platforms, packages and libraries** I get this error

Error: Traceback (most recent call last):
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/commands/update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py", line 385, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 132, in update
    self.cleanup_packages(p.packages.keys())
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 144, in cleanup_packages
    p = PlatformFactory.newPlatform(manifest['__pkg_dir'])
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 250, in newPlatform
    _instance = platform_cls(join(platform_dir, "platform.json"))
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 460, in __init__
    self._manifest = util.load_json(manifest_path)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 169, in load_json
    with open(file_path, "r") as f:
IOError: [Errno 2] No such file or directory: '/Users/djmanas/.platformio/platforms/esp-idf/platform.json'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

#### Steps to Reproduce

1. I had upgraded to new version of plugin, reloaded
2. pressed cmd+shift+p
3. selected Platformio: Update installed platforms, packages and libraries

### Actual Results
```
Error: Traceback (most recent call last):
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/commands/update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py", line 385, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 132, in update
    self.cleanup_packages(p.packages.keys())
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 144, in cleanup_packages
    p = PlatformFactory.newPlatform(manifest['__pkg_dir'])
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 250, in newPlatform
    _instance = platform_cls(join(platform_dir, "platform.json"))
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 460, in __init__
    self._manifest = util.load_json(manifest_path)
  File "/Users/djmanas/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 169, in load_json
    with open(file_path, "r") as f:
IOError: [Errno 2] No such file or directory: '/Users/djmanas/.platformio/platforms/esp-idf/platform.json'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Expected Results
Platforms, libs, etc updated

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```N/A
```

**Source file to reproduce issue:**
```N/A
```

### Additional info
I had messed with ESPIDF framework to let latest json, but version from git. It didn’t work out so I had deleted platform-espidf folder, opened ESP32 project and gave compile to let PIO download correct verison of this package. After downloading and restarting of VSCode it is still the same.

ivankravets, 2018-04-29:
Where did you get `/Users/djmanas/.platformio/platforms/esp-idf `?

DJManas, 2018-04-30:
Now I see it. When you had told me I can use github repo for ESP8266 I tried it with esp32 platform to get C code to compile:

```
[env:esp32dev]
platform = https://github.com/espressif/esp-idf.git
board = esp32dev
framework = espidf
```

My fault. Sorry :-(

closed 2018-04-30 by ivankravets