platformio/platformio-core #1609: PlatformIO dependency does not work. Debian Linux.

AsafFisher, 2018-05-05:
-----------------

### Configuration
Debian Linux

PlatformIO, version 3.5.2

### Description of problem
I am trying to build my project, all the dependencies are in my platformio.ini
It gives me this error when I compile the project:
src/AccountList.h:2:30: fatal error: Adafruit_SSD1306.h: No such file or directory

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
lib_deps =
  13
  https://github.com/AsafFisher/Adafruit_SSD1306.git ; Adjusted id=135 to work with my 128x64 setup

```

### Additional info
```
Compiling .pioenvs/esp32dev/src/AccountList.cpp.o
In file included from src/AccountList.cpp:1:0:
src/AccountList.h:2:30: fatal error: Adafruit_SSD1306.h: No such file or directory
 
**************************************************************************
* Looking for Adafruit_SSD1306.h dependency? Check our library registry!
*
```

AsafFisher, 2018-05-05:
I did use pio update/upgrade everything like that

AsafFisher, 2018-05-05:
More info
```
[Fri May  4 20:41:58 2018] Processing esp32dev (platform: https://github.com/platformio/plat

form-espressif32.git#feature/stage; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'https://github.com/AsafFisher/Adafruit_SSD
1306.git ; Adjusted id'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'https://github.com/AsafFish
er/Adafruit_SSD1306.git ; Adjusted id'}` is a built-in library (included in framework, SDK).
 E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 (Stage) > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
```

ivankravets, 2018-05-08:
Please move this comment form a new line " ; Adjusted id=135..."

closed 2018-05-08 by ivankravets

AsafFisher, 2018-05-08:
Are you saying that the comment being in the same line is the problem?

ivankravets, 2018-05-08:
Yes, that is not valid INI syntax. See https://docs.python.org/2/library/configparser.html

> Lines beginning with '#' or ';' are ignored and may be used to provide comments.