platformio/platformio-core #1617: Home: Libraries: Could not update library

davidbitton, 2018-05-09:
PIO Core Call Error: Library Storage: /Users/bittond/GoogleDrive/src/Jeti_Omnibus-Sensor/.piolibdeps
Updating ArduinoMenu                     @ 30084f0        [66a8827]
git version 2.15.1 (Apple Git-101)
Updating 30084f0..66a8827

From https://github.com/neu-rah/ArduinoMenu
   30084f0..66a8827  master     -> origin/master
 * [new tag]         v4.5.0     -> v4.5.0
 * [new tag]         4.4.6      -> 4.4.6
 * [new tag]         v4.4.7     -> v4.4.7
 * [new tag]         v4.4.9     -> v4.4.9
error: Your local changes to the following files would be overwritten by merge:
	src/io.h
Please commit your changes or stash them before you merge.
Aborting
Error: VCS: Could not process command ['git', 'pull', '--recurse-submodules']

ivankravets, 2018-05-09:
Have you did some changes to cloned library

davidbitton, 2018-05-09:
I believe I did which I think are now incorporated into the lib. How can I revert the local changes so I can pull down the latest?