platformio/platformio-core #1621: Mac OS 10.11.6: pio home aborts with: No module named platform

michaelrommel, 2018-05-10:
What kind of issue is this? -> PlatformIO Core.
------------------------------------------------------------------

### Configuration

**Operating system**:
Mac OS X 10.11.6

**PlatformIO Version** (`platformio --version`):
platformio --version
PlatformIO, version 3.5.3b3
pio --version
PlatformIO, version 3.5.3b3

which platformio
/usr/local/bin/platformio
ls -la /usr/local/bin/platf* /usr/local/bin/pio
-rwxr-xr-x  1 rommel  admin  243 May 10 16:19 /usr/local/bin/pio
-rwxr-xr-x  1 rommel  admin  243 May 10 16:19 /usr/local/bin/platformio

pio update
Updating tool-scons                      @ 2.20501.4      [Up-to-date]
Updating contrib-pysite                  @ 0.2.0          [Up-to-date]
Updating tool-pioplus                    @ 1.3.0          [Up-to-date]

Platform Manager

Platform Atmel AVR

Updating atmelavr                        @ 1.8.2          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]

Platform Atmel SAM

Updating atmelsam                        @ 3.1.0          [Up-to-date]
Updating framework-arduinosam            @ 2.10618.1      [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.40804.0      [Up-to-date]

Library Manager

pio upgrade
You're up-to-date!
PlatformIO 3.5.3b3 is currently the newest version available.

pio platforms list
atmelavr ~ Atmel AVR
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Frameworks: arduino, simba
Packages: toolchain-atmelavr, framework-arduinoavr, framework-simba, tool-avrdude, tool-micronucleus
Version: 1.8.2

atmelsam ~ Atmel SAM
Atmel | SMART offers Flash- based ARM products based on the ARM Cortex-M0+, Cortex-M3 and Cortex-M4 architectures, ranging from 8KB to 2MB of Flash including a rich peripheral and feature mix.

Home: http://platformio.org/platforms/atmelsam
Frameworks: arduino, mbed, simba
Packages: toolchain-gccarmnoneeabi, framework-arduinosam, framework-simba, tool-openocd, framework-mbed, tool-avrdude, tool-bossac
Version: 3.1.0

pio run
[Thu May 10 16:47:48 2018] Processing due (platform: atmelsam; board: due; framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel SAM > Arduino Due (Programming Port)
SYSTEM: AT91SAM3X8E 84MHz 32KB RAM (512KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 3 compatible libraries
Scanning dependencies...
No dependencies
Calculating size .pioenvs/due/firmware.elf
text	   data	    bss	    dec	    hex	filename
24536	      0	   2600	  27136	   6a00	.pioenvs/due/firmware.elf
[SUCCESS] Took 3.16 seconds

### Description of problem

The base system runs, source files can be compiled, platforms could be installed. However, trying to setup the local listener bails. I had the same issue, when installing everything via the Atom IDE, same errors, I uninstalled everything, to get back to basics.

pio home
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 19, in init cmd_debug
ImportError: No module named platform

pio home --port 8888
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 19, in init cmd_debug
ImportError: No module named platform

#### Steps to Reproduce

1. Removed Atom IDE, removed ~/.platformio
2. pip uninstall platformio
3. pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
4. platformio update
5. platformio upgrade
6. platformio platform install atmelsam
7. platformio platform install atmelavr
8. pio home

### Actual Results
Error is thrown.

### Expected Results
Local listener is opened.

ivankravets, 2018-05-13:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

michaelrommel, 2018-05-13:
$ echo $PATH
/usr/local/opt/gnu-sed/libexec/gnubin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Volumes/Data/Users/rommel/bin:/Volumes/Data/Software/michael/mongodb/bin:/Applications/Atom.app/Contents/Frameworks/Atom Helper.app/Contents/MacOS/Atom Helper:/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/
$ echo $PYTHONPATH

$ python --version
Python 2.7.15
l$ pip --version
pip 10.0.1 from /usr/local/lib/python2.7/site-packages/pip (python 2.7)
$ pip list
Package          Version
---------------- ---------
bottle           0.12.13
certifi          2018.4.16
chardet          3.0.4
click            5.1
colorama         0.3.9
idna             2.6
lockfile         0.12.2
pip              10.0.1
platformio       3.5.3b3
pyserial         3.4
requests         2.18.4
semantic-version 2.6.0
setuptools       39.1.0
urllib3          1.22
wheel            0.31.0

ivankravets, 2018-05-18:
Please don't install PIO Core manually. Try to remove all PIO Cores from a system - http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Then, remove `~/.platformio` folder and start IDE

michaelrommel, 2018-05-20:
Well, I tried that before. several times. It always ended up with the same error, which "is already reported" (https://github.com/platformio/platformio-atom-ide/issues/1360) but closed. All fixes in this thread did not work, basically the thread started out as Mac OS X reported, but later it switched to Windows and the fixes do not work.

I tried to do the base install without IDE to provide you with a less complex system to deal with and make the error tracking easier.

But as you wished, I tried it again and came up with the same error. Please see attached screenshots after installation and after restart.

Here are the now active command outputs from pip list, the rest remained unchanged:

$ pip list
Package          Version
bottle           0.12.13
certifi          2018.4.16
chardet          3.0.4
click            5.1
colorama         0.3.9
idna             2.6
lockfile         0.12.2
pip              10.0.1
pyserial         3.4
requests         2.18.4
semantic-version 2.6.0
setuptools       39.1.0
urllib3          1.22
wheel            0.31.0

Atom is at its latest version, all other packages have also been updated. Still the error persists.

[image]
[image]

ivankravets, 2018-06-16:
Do you still have this "Mac OS X 10.11.6" machine? I would like to ask you to test some code. If it works, I'll work on the fix for this issue.

michaelrommel, 2018-06-18:
Hello, yes, I still work on that machine. I have now switched to Arduino-Makefile for my needs of a decent editor/make workflow. But I can help you and test some code.

ivankravets, 2018-06-18:
Please open PlatformIO IDE for VSCode and type in terminal (you should see an icon on the bottom toolbar)
```
pip list
python -c "from platformio.commands import platform; print platform, dir(platform)"
```

michaelrommel, 2018-06-18:
```
$ pip list
Package          Version
---------------- ---------
bottle           0.12.13
certifi          2018.4.16
chardet          3.0.4
click            5.1
colorama         0.3.9
idna             2.6
lockfile         0.12.2
pip              10.0.1
platformio       3.6.0a2

pyserial         3.4
requests         2.18.4
semantic-version 2.6.0
setuptools       39.2.0
urllib3          1.22
wheel            0.31.1

$ python -c "from platformio.commands import platform; print platform, dir(platform)"
<module 'platformio.commands.platform' from '/Volumes/Data/Users/rommel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.pyc'> ['PlatformFactory', 'Pl
atformManager', '__builtins__', '__doc__', '__file__', '__name__', '__package__', '_get_installed_platform_data', '_get_platform_data', '_get_registry_platform_data', '_get_re
gistry_platforms', '_original_version', '_print_platforms', 'app', 'cli', 'click', 'dirname', 'exception', 'isdir', 'json', 'platform_frameworks', 'platform_install', 'platfor
m_list', 'platform_search', 'platform_show', 'platform_uninstall', 'platform_update', 'print_boards', 'util']
```

ivankravets, 2018-06-21:
Could you repeat the same command from your System Terminal? Without IDE.
```
python -c "from platformio.commands import platform; print platform, dir(platform)"
```

michaelrommel, 2018-06-21:
Hi Ivan,
here is the result:
```
$ python -c "from platformio.commands import platform; print platform, dir(platform)"
<module 'platformio.commands.platform' from '/Volumes/Data/Users/rommel/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.pyc'> ['PlatformFactory', 'PlatformManager', '__builtins__', '__doc__', '__file__', '__name__', '__package__', '_get_installed_platform_data', '_get_platform_data', '_get_registry_platform_data', '_get_registry_platforms', '_original_version', '_print_platforms', 'app', 'cli', 'click', 'dirname', 'exception', 'isdir', 'json', 'platform_frameworks', 'platform_install', 'platform_list', 'platform_search', 'platform_show', 'platform_uninstall', 'platform_update', 'print_boards', 'util']
``` 
Thank you for your investigation! I saw that also ESP32 modules are supported on PlatformIO - I still think it might be worth a try...

  Michael.

ivankravets, 2018-06-21:
Please replace `~/.platformio/packages/tool-piplus` with the patached files in attach.

Then, open PIO IDE Terminal and type `pio account show` or other PIO Plus Core command. Please provide an output here.

[tool-pioplus.zip](https://github.com/platformio/platformio-core/files/2124761/tool-pioplus.zip)

michaelrommel, 2018-06-21:
Hi,

it outputs:

```
$ pio account show
['/Volumes/Data/Users/rommel/.platformio/packages/contrib-pysite', '/Library/Python/2.7/site-packages', '/Volumes/Data/Users/rommel/.platformio/packages/tool-pioplus', '/Libra
ry/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg', '/Library/Python/2.7/site-packages', '/Library/Python/2.7/site-packages/PyVISA-1.9.dev0-py2.7.egg', '/Library/Python/2.7/
site-packages/enum34-1.1.2-py2.7.egg', '/Volumes/Data/Users/rommel/.platformio/penv/lib/python2.7/site-packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib
/python27.zip', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin',
 '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-script
packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/Sys
tem/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/System/Lib
rary/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC', '/Library/Python/2.7/site-packages', '/Volumes/Data/Users/rommel/.platformio/packages/contrib-pysite']
Traceback (most recent call last):
  File "main.py", line 13, in init main
ImportError: cannot import name platform
$
```

michaelrommel, 2018-06-21:
It still gives an error - this is from inside the Atom IDE:

```
$ pio account show
['/Volumes/Data/Users/rommel/.platformio/packages/contrib-pysite', '/Library/Python/2.7/site-packages', '/Volumes/Data/Users/rommel/.platformio/packages/tool-pioplus', '/Libra

ry/Python/2.7/site-packages/esptool-0.1.0-py2.7.egg', '/Library/Python/2.7/site-packages', '/Library/Python/2.7/site-packages/PyVISA-1.9.dev0-py2.7.egg', '/Library/Python/2.7/
site-packages/enum34-1.1.2-py2.7.egg', '/Volumes/Data/Users/rommel/.platformio/penv/lib/python2.7/site-packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib
/python27.zip', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin',
 '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-script
packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/Sys
tem/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/System/Lib
rary/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC', '/Library/Python/2.7/site-packages', '/Volumes/Data/Users/rommel/.platformio/packages/contrib-pysite']
13 <module 'platformio.app' from '/Library/Python/2.7/site-packages/platformio/app.pyc'>
Traceback (most recent call last):
  File "main.py", line 15, in init main
ImportError: cannot import name platform
```

ivankravets, 2018-06-22:
> /Library/Python/2.7/site-packages/platformio/app.py

You have multiple PIO Cores in a system. I already asked you here (https://github.com/platformio/platformio-core/issues/1621#issuecomment-390052467) to delete them. What is more, one of them is totally broken and does not contain all files.

Please navigate to `/Library/Python/2.7/site-packages/` folder and remove all folders which contain "platformio" in the name. It should work now.

michaelrommel, 2018-06-22:
Hello Ivan,

thank you! Yes, I did try to follow your instructions and when I did a `brew list` or  a `pip list`, no other installation showed. But I now think I understood where the problem lies - by putting .platformio to the front in my path, the pip from platformio was used. But the other pip in the global install would have listed the erroneous module still! I didn't know that, maybe this is an description in the FAQ worth, to let people see, which pip they are using and each pip seems to look into different paths. (I didn't know enough python to track this down...)

```
$ /usr/local/bin/pip list
Package          Version  
---------------- ---------
bottle           0.12.13  
certifi          2018.4.16
chardet          3.0.4    
click            5.1      
colorama         0.3.9    
idna             2.6      
lockfile         0.12.2   
pip              10.0.1   
pyserial         3.4      
requests         2.18.4   
semantic-version 2.6.0    
setuptools       39.1.0   
urllib3          1.22     
wheel            0.31.0   

$ pip list
Package          Version  
---------------- ---------
bottle           0.12.13  
certifi          2018.4.16
chardet          3.0.4    
click            5.1      
colorama         0.3.9    
idna             2.6      
lockfile         0.12.2   
pip              10.0.1   
platformio       3.6.0a3  
pyserial         3.4      
requests         2.18.4   
semantic-version 2.6.0    
setuptools       39.2.0   
urllib3          1.22     
wheel            0.31.1   

$ which pip
/Volumes/Data/Users/rommel/.platformio/penv/bin//pip
```

The IDE in Atom works now! Thank you very much!

  Michael.

michaelrommel, 2018-06-22:
Issue closed, there was an undetected second `pip` installation of platformio in the system.

closed 2018-06-22 by michaelrommel

ivankravets, 2018-06-22:
Happy coding with PlatformIO! 😊