platformio/platformio-core #1638: Can not use PIO Plus on nixOS

ivankravets, 2018-05-26:
Moved from https://github.com/platformio/platformio-vscode-ide/issues/105

/cc @ShiroUsagi-san

mogorman, 2018-05-26:
I am on nixos 18.03 which has 3.5.1 packaged and it seemed to work for me.
```
$pio home
*****************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
*****************************************************************************************************************************************************************************************************************

CorePackageManager: Installing tool-pioplus @ >=0.13.3,<2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-pioplus @ 1.3.1 has been successfully installed!
CorePackageManager: Installing contrib-pysite @ >=0.1.2,<2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
contrib-pysite @ 0.2.0 has been successfully installed!
PlatformIO Plus (https://pioplus.com) v1.3.1
CorePackageManager: Installing contrib-piohome @ >=0.6.1,<2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
contrib-piohome @ 0.9.5 has been successfully installed!

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

^C

Aborted!

[mog@toadman:~]$ pio --version
PlatformIO, version 3.5.1

```

mogorman, 2018-05-26:
actually i checked on my 18.09 box and it worked there as well.
```
$ pio home
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 3.5.2!

*****************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
*****************************************************************************************************************************************************************************************************************

CorePackageManager: Installing tool-pioplus @ >=0.14.5,<2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-pioplus @ 1.3.1 has been successfully installed!
CorePackageManager: Installing contrib-pysite @ >=0.1.5,<2
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
contrib-pysite @ 0.2.0 has been successfully installed!
PlatformIO Plus (https://pioplus.com) v1.3.1
CorePackageManager: Installing contrib-piohome @ >=0.7.1,<2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
contrib-piohome @ 0.9.5 has been successfully installed!

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

^C

[mog@zaphod:~]$ Aborted!
pio home^C

[mog@zaphod:~]$ pio --version
PlatformIO, version 3.5.2

```

ivankravets, 2018-05-27:
@ShiroUsagi-san could you re-test? Does it work with the latest version? What is your NixOS version?

ShiroUsagi-san, 2018-05-27:
@ivankravets actually, my nixos installation has been broken so I had to change to archlinux, so I dont have nixos anymore :( . Sorry

closed 2018-05-27 by ivankravets