platformio/platformio-core #1641: Build-error for AVR-GCC because MCU and F_CPU settings not correctly passed

martinjaeger, 2018-05-28:
What kind of issue is this?

- [x] Development Platform or Board (maybe: platform-atmelavr)

- [x] PlatformIO Core.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 7

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.3rc1

### Description of problem

With the new platformio.ini options board_build.mcu (instead of board_mcu) and board_build.f_cpu (instead of board_f_cpu) my Atmel AVR project does not compile anymore.

It seems like the environment variables $BOARD_MCU and $BOARD_F_CPU are not correctly passed to main.py of platform-atmelavr, so I believe it's an error in the platformio-core.

pio run -v output:
```
avr-gcc -o .pioenvs\uno\src\main.o -c -std=gnu11 -fno-fat-lto-objects -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu= -DF_CPU= -DPLATFORMIO=30503 -D__AVR_ATmega32U4__ -Isrc src\main.c
```

#### Steps to Reproduce

1. Compile project for Atmel AVR platform

### Actual Results

Compiler and linker errors because -mmcu is not specified:
```
avr-gcc: error: missing device or architecture after '-mmcu='
```

### Expected Results

Clean compilation (worked with previous platformio versions (don't know which version).

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:board]
platform = atmelavr
board_build.mcu = atmega32u4
board_build.f_cpu = 16000000L
upload_protocol = usbtiny
upload_speed = 19200
```

ivankravets, 2018-05-30:
Thanks! Please re-test with `pio upgrade --dev` version.

closed 2018-05-30 by ivankravets

martinjaeger, 2018-05-30:
Perfect. Now it works. Thank you!

commit b213a302 (2018-05-30): Handle _MCU and _F_CPU variables for AVR native // Resolve #1641