platformio/platformio-core #1642: Support

OscarGarciaF, 2018-05-28:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 3.5.3rc1

### Description of problem

I have modified File, FileSystem and FatFileSystem files of the mbed framework to add the function truncate and it works as intended. The problem i'm working with a team on a git and i would like them to be able to access this customs files without having to modify their frameworks files so i added the entire FileSystem folder (with the customs files) to the lib folder in the project and restore the original files in the .platformio folder.

platformio is supposed to be prefer the files in the lib folder over the ones in the installation, however it appears to be not working as intended as it's including the original files instead of the customs one (causing a compile error since it cannot find my member function).

I have posted in the community folder [here](https://community.platformio.org/t/superseding-framework-libraries/4209)

ivankravets, 2018-05-29:
Resolved in https://community.platformio.org/t/superseding-framework-libraries/4209/6?u=ivankravets

closed 2018-05-29 by ivankravets

OscarGarciaF, 2018-05-29:
Still not solved