platformio/platformio-core #1658: Home: Libraries: Stats

seksity, 2018-06-05:
PIO Core Call Error: Error: [API] Could not connect to PlatformIO API Service. Please try later.

tbrasser, 2018-06-05:
Yeah, seems as if *.platformio.org is offline. I can't build anything now.....

SuperMarcus, 2018-06-05:
Yes, I also get a connection refused when trying to access the website directly, and nothings builds without `api.platformio.org`.

```
* Rebuilt URL to: api.platformio.org/
*   Trying 193.222.52.25...
* TCP_NODELAY set
* Connection failed
* connect to 193.222.52.25 port 80 failed: Connection refused
* Failed to connect to api.platformio.org port 80: Connection refused
* Closing connection 0
curl: (7) Failed to connect to api.platformio.org port 80: Connection refused
```

```
* Rebuilt URL to: https://api.platformio.org/
*   Trying 193.222.52.25...
* TCP_NODELAY set
* Connection failed
* connect to 193.222.52.25 port 443 failed: Connection refused
* Failed to connect to api.platformio.org port 443: Connection refused
* Closing connection 0
curl: (7) Failed to connect to api.platformio.org port 443: Connection refused
```

yozik04, 2018-06-05:
Why we should care if some site is down. It should have an option to compile without internet.

ivankravets, 2018-06-05:
@yozik04 you just need internet/PIO LIbrary Registry when you build your project first time and it depends on extra libraries from PlatformIO Registry.

ivankravets, 2018-06-05:
Sorry for the issue. The first unexpected downtime for the last 3 years. We will investigate this issue in the logs.

Now everything works well.

yozik04, 2018-06-05:
@ivankravets Actually I am not building the project for the first time. I have all libs locally. If I disconnect from wifi then build succeeds.

ivankravets, 2018-06-05:
>  If I disconnect from wifi then build succeeds.

Yes, that is correct behavior. PlatformIO SHOULD work in offline mode without any problems. Please ping me if something is wrong.

OttoWinter, 2018-06-05:
The problem was that if platformio detected that a network connection existed (through some other method I presume), but api.platformio.org was unreachable, the build failed.

This was what was removing the ability to build in the brief time period before. Disconnecting WiFi during that time had the correct result of just printing a warning and trying to build anyway. So if internet connection exists, but API not reachable, the build fails even though the libraries are installed locally.

yozik04, 2018-06-05:
@OttoWinter is right. If feels like a place to do some improvements.

ivankravets, 2018-06-05:
> So if internet connection exists, but API not reachable, the build fails even though the libraries are installed locally.

Thanks! I'll retest that later. I've just reopened this issue.

ivankravets, 2018-06-15:
I can't reproduce this issue. Do you have a firewall? Could you try to block `api.platformio.org` and re-test?

ivankravets, 2018-06-16:
Please reopen if you still need help

closed 2018-06-16 by ivankravets

DrMarkyMarc, 2019-04-02:
same issue after changer dns and having my hassio ip change wifi->wlan.

sjalife, 2019-07-01:
It appears to be down again.

I keep getting:

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f19e51c7c90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)

and after 5 attempts:

Error: [API] Could not connect to PlatformIO API Service. Please try later.

DeeBeeKay, 2019-07-02:
I don't have anything to add other than, I can't build anything at the moment either, because of this issue, whatever it is.

ivankravets, 2019-07-02:
Do you have access to https://api.platformio.org/lib/stats ?

DeeBeeKay, 2019-07-02:
Thank you for your rapid response. It is mysteriously working once again, having been dead for about 4 hours. I'm trying to think what I can have done to fix it, but nothing particularly makes sense.

Things I tried:

- Disabled my pihole. No effect.
- I read above that disabling the network interface may force it to proceed using local files, so, disabled the WAN connection in my router since I'm logged into a pi that's elsewhere on the local network. No appreciable effect, I just got yelled at about there being no internet connection.
- Re-enabled the WAN. Cursed as realised killing the WAN had also killed download of new version of FL Studio. I am an idiot.
- Stumped off to workshop to fetch a tool.
- Returned, and it's working.

Sorry to have bothered you.

sjalife, 2019-07-03:
So, I still have this issue.

I am running Haas from within a Docker instance on Ubuntu. And ESPHome
within HAAS.io.

The Ubuntu machine has access to the internet, and does resolve DNS queries
without issue.

I have checked the DNS entries for resolve, for the docker config on the
host and for the docker image itself as per other solutions found on the
forum (and I know it works because HA has internet access, downloads
add-ons, and updates OK).

Yet still JUST the ESPHome add-on does not resolve DNS requests and as
such, I cannot build a thing.

Any ideas?

ivankravets, 2019-07-04:
Could you provide an output from 
```
traceroute api.platformio.org
```

sjalife, 2019-07-07:
Sure. 

When running a traceroute (or a ping) from HAAS.io I get the "traceroute: bad address 'api.platformio.org' " error.

ping to 8.8.8.8 works OK. 

I resolved on MY LAPTOP api.platformio.org to 193.222.52.25. 
Pinged  from HASS.io to it without issues.

I still cannot resolve any DNS addresses from within HASS.io, yet it works from the HA instance and from the linux computer OK.

Now, a traceroute to 193.222.52.25 shows the correct routing order and internal hop count for my network.

See attached screenshots for all above.
(Sorry in advance for the first couple of typos on the addresses and commands)

[image]
[image]
[image]

chriscrowe, 2019-08-17:
Just in case anyone else comes across this-- my issue was that my network firewall was GeoIP blocking IPs in Ukraine (along with a bunch of other countries), and it seems platformio's hosting is in Ukraine. Removed Ukraine from the blocklist and all is well now.

sjwright, 2020-02-27:
Platformio is down again...

germanotero, 2020-06-04:
and again

ivankravets, 2020-06-04:
https://status.platformio.org/

There are no issues with our APIs.

sag3ll0, 2020-06-04:
ping cannot access
[image]

ivankravets, 2020-06-04:
This is ok that you can’t ping PlatformIO.org. We disabled ICMP packets for security reason. Try “curl https://platformio.org/“. It should work.

sag3ll0, 2020-06-04:
> It should work.

yes, its work
but i got error while try compile firmware
[image]

111lisu, 2020-06-05:
```
INFO Reading configuration /config/esphome/inkubator_s.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/inkubator_s
Processing inkubator_s (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ArduinoJson-esphomelib library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
Compiling /data/inkubator_s/.pioenvs/inkubator_s/src/esphome/components/api/api_connection.cpp.o
Compiling /data/inkubator_s/.pioenvs/inkubator_s/src/esphome/components/api/api_server.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/inkubator_s/.pioenvs/inkubator_s/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/inkubator_s/.pioenvs/inkubator_s/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [FAILED] Took 5.98 seconds ==========================

#############################################################################

root@15ef4d2f-esphome:/# platformio lib search "header:ESPAsyncTCP.h"
Error: You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.

root@15ef4d2f-esphome:/# telnet api.platformio.org 80
Trying 31.28.1.238...
Connected to nm1.platformio.org.
Escape character is '^]'.                               Running 

root@15ef4d2f-esphome:/# telnet api.platformio.org 443
Trying 31.28.1.238...
Connected to nm1.platformio.org.
Escape character is '^]'.                              Running
```