platformio/platformio-core #1659: PlatformIO website is down

OttoWinter, 2018-06-05:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS & debian
**PlatformIO Version** (`platformio --version`):3.5.2
### Description of problem

The platformIO website is down, thus causing all builds to fail unless the internet connection is actively disabled.

#### Steps to Reproduce

1. Try to compile a platformIO project
2. Observe these messages if there are some library dependencies:

```bash
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ArduinoOTA%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x10ee7f810>: Failed to establish a new connection: [Errno 61] Connection refused',)) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ArduinoOTA%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x10efb3350>: Failed to establish a new connection: [Errno 61] Connection refused',)) (incremented retry: max=5, total=2)
```

3. Try disabling WiFi, now the project compiles with warnings - but this will only work for users with existing projects:

```bash
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.
You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchain.
```

### Actual Results

### Expected Results

platformio.org should be reachable

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

Also seems like the main website (not the API) is not reachable too: http://platformio.org/

OttoWinter, 2018-06-05:
Whoops .. duplicate of #1658 - Close this if needed

OttoWinter, 2018-06-05:
Update: community forums are also down: https://community.platformio.org/
but the docs are still working: http://docs.platformio.org/en/latest/faq.html (probably distributed through read the docs, if I see correctly)

Leads me to believe the platformio servers are unreachable ATM - the DNS records look fine to me

OttoWinter, 2018-06-05:
It's up again 😀

Thanks guys!

sjalife, 2019-07-01:
It appears to be down again.

I keep getting:

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f19e51c7c90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)

and after 5 attempts:

Error: [API] Could not connect to PlatformIO API Service. Please try later.

DirkTas67, 2019-12-23:
and again today 23 December 2019

ModischFabrications, 2019-12-25:
and today, 25 December 2019.

ivankravets, 2019-12-25:
Sorry for the issue. There are hardware issues (motherboard) with our dedicated server located in Ukraine. The server reboots without a reason and there is nothing interesting in logs. On Monday, the server reported that there is a problem with 2 banks of RAM. We removed them. However, there are new problems today.

We've already ordered a new server but it takes 1 business day :( Today is Christmas in Netherland.

P.S: We also stopped all services on the current server and kept only PlatformIo Registry API and   Community Forums. The PIO Plus does not work, it requires a lot of networking. 

We will keep you updated on this thread.

Merry Christmas!

ivankravets, 2019-12-25:
@ModischFabrications The services are online now.

ModischFabrications, 2019-12-25:
Thanks for the info, enjoy your holiday! 🎄

brandond, 2019-12-26:
It's down again :(

nurikk, 2019-12-26:
> @ModischFabrications The services are online now.

it's down

nurikk, 2019-12-26:
@ivankravets it's down

red-scorp, 2019-12-26:
**platformio.org** is unreachable from Germany. I hope it will be available soon.
Thanks for your working on holidays.
Merry Christmas! 🎄

antlob76, 2019-12-26:
platformio.org is unreachable from Spain as well. Please tell us when it is back online again.  Hope the issue gets fixed.

proddy, 2019-12-26:
same in NL

pawel-manu, 2019-12-26:
Same in PL :(

ivankravets, 2019-12-26:
We are restoring yesterday's backup in a new DC. Our previous dedicated server which has worked since 2014 stopped booting today. Motherboard issues. We tried to resolve this issue yesterday. However, no one works on the 25-26 of Dec in Europe, we can't order normal server. So, moving everything to VPS (temporary).

d3dx9, 2019-12-26:
Is there any workaround where I can place the packages manually while the downtime?

ivankravets, 2019-12-26:
Yes, use direct GIT url in `lib_deps`

d3dx9, 2019-12-26:
Thank you, it worked. :)

SJ-Innovation, 2019-12-26:
Of all the times for server issues! Christmas! Best of luck, thank you for everything, and merry christmas @ivankravets (et al)!

JBM-fr, 2019-12-26:
> Thank you, it worked. :)

Hello ... can you explain me how to do this ?
Thank you ...

ivankravets, 2019-12-26:
API is online now https://platformio.org/lib. Other services are coming soon.

red-scorp, 2019-12-26:
> API is online now https://platformio.org/lib. Other services are coming soon.
Thanks, it helped a lot.

ivankravets, 2019-12-27:
All services were restored yesterday.

Happy New Year!

closed 2019-12-27 by ivankravets