platformio/platformio-core #1674: Clion code autocompletion (import Arduino.h)

vitt0ri0, 2018-06-12:
Hello! 

I cannot get to work code autocompletion in Clion. 

I have followed this guide (http://docs.platformio.org/en/latest/ide/clion.html) and installed PIO Core, then I have created a new blink project. I have managed it to work from command line. 
(Just created a main.cpp with "blink" content and executed pio run -t upload. It works nice). 

However I need autocompletion in CLion. I have added a line 
`add_executable(myproject src/main.cpp)` 
into CMakeLists.txt, but i doesn't help.

ivankravets, 2018-06-15:
Could you try to repeat `pio init --ide clion` when you modify `src` or `lib` folder? Restart CLion, it should work.

Techzune, 2018-12-13:
Currently experiencing this issue. Running PLATFORMIO_REBUILD_PROJECT_INDEX causes CLion 2018.3.1 to forget `pio` and `platformio` commands. Running `pio init --ide clion` does re-init the project, but code completion still throws errors such as Arduino.h being nonexistent. 😢

ivankravets, 2018-12-13:
Fixed in https://github.com/platformio/platformio-core/commit/9f4dde4b5ea99ed25ec0dc530039bf84b800dd86

Please run in CLI `pio upgrade --dev` and re-init project. Does `PLATFORMIO_REBUILD_PROJECT_INDEX` work now?

closed 2018-12-13 by ivankravets