platformio/platformio-core #1676: Getting multiple definitions trying to compile Blynk ESP32 SSL example

treii28, 2018-06-12:
What kind of issue is this?

------------------------------------------------------------------

### Configuration

**Operating system**:
linux

**PlatformIO Version** (`platformio --version`):
3.5.3

### Description of problem
Trying to compile bare bones example sketch from blynk

#### Steps to Reproduce
installed Blynk library (415)
initialized project for board esp32dev with ide clion
copied Esp32 SSL example to src/main.cpp
added include for Arduino.h at top of example
ran platformio run

### Actual Results

```
> pio run
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
[Tue Jun 12 12:23:07 2018] Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
warnings.warn(warning, RequestsDependencyWarning)
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
DEBUG: CURRENT(minimodule) EXTERNAL(minimodule, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h)
Converting ESP32_WiFi_SSL.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Blynk> v0.5.3
|   |-- <WiFiClientSecure> v1.0
|   |   |-- <WiFi> v1.0
|   |-- <WiFi> v1.0
|-- <WiFi> v1.0
|-- <WiFiClientSecure> v1.0
|   |-- <WiFi> v1.0
Compiling .pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o
Compiling .pioenvs/esp32dev/lib712/Blynk_ID415/utility/BlynkDebug.cpp.o
Compiling .pioenvs/esp32dev/lib712/Blynk_ID415/utility/BlynkHandlers.cpp.o
Compiling .pioenvs/esp32dev/lib712/Blynk_ID415/utility/BlynkTimer.cpp.o
Compiling .pioenvs/esp32dev/lib712/Blynk_ID415/utility/utility.cpp.o
Archiving .pioenvs/esp32dev/lib712/libBlynk_ID415.a
Indexing .pioenvs/esp32dev/lib712/libBlynk_ID415.a
Linking .pioenvs/esp32dev/firmware.elf
.pioenvs/esp32dev/src/main.cpp.o: In function `blynk_dbg_print(char const*, ...)':
main.cpp:(.text._Z15blynk_dbg_printPKcz+0x0): multiple definition of `blynk_dbg_print(char const*, ...)'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:ESP32_WiFi_SSL.ino.cpp:(.text._Z15blynk_dbg_printPKcz+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o:(.data.pass+0x0): multiple definition of `pass'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:(.data.pass+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o:(.data.ssid+0x0): multiple definition of `ssid'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:(.data.ssid+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o:(.bss.Blynk+0x0): multiple definition of `Blynk'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:(.bss.Blynk+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o:(.data.auth+0x0): multiple definition of `auth'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:(.data.auth+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o: In function `setup()':
main.cpp:(.text._Z5setupv+0x0): multiple definition of `setup()'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:ESP32_WiFi_SSL.ino.cpp:(.text._Z5setupv+0x0): first defined here
.pioenvs/esp32dev/src/main.cpp.o: In function `loop()':
main.cpp:(.text._Z4loopv+0x0): multiple definition of `loop()'
.pioenvs/esp32dev/src/ESP32_WiFi_SSL.ino.cpp.o:ESP32_WiFi_SSL.ino.cpp:(.text._Z4loopv+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32dev/firmware.elf] Error 1
========================== [ERROR] Took 3.83 seconds ==========================

```
### Expected Results
Success

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#ifndef Arduino_h
#include <Arduino.h>
#endif
/*************************************************************
  Download latest Blynk library here:
    https://github.com/blynkkk/blynk-library/releases/latest

  Blynk is a platform with iOS and Android apps to control
  Arduino, Raspberry Pi and the likes over the Internet.
  You can easily build graphic interfaces for all your
  projects by simply dragging and dropping widgets.

    Downloads, docs, tutorials: http://www.blynk.cc
    Sketch generator:           http://examples.blynk.cc
    Blynk community:            http://community.blynk.cc
    Follow us:                  http://www.fb.com/blynkapp
                                http://twitter.com/blynk_app

  Blynk library is licensed under MIT license
  This example code is in public domain.

 *************************************************************
  This example runs directly on ESP32 chip.

  Note: This requires ESP32 support package:
    https://github.com/espressif/arduino-esp32

  Please be sure to select the right ESP32 module
  in the Tools -> Board menu!

  Change WiFi ssid, pass, and Blynk auth token to run :)
  Feel free to apply it to any other example. It's simple!
 *************************************************************/

/* Comment this out to disable prints and save space */
#define BLYNK_PRINT Serial

#ifndef WiFi_h
#include <WiFi.h>
#endif

#ifndef WiFiClientSecure_h
#include <WiFiClientSecure.h>
#endif

#ifndef BlynkSimpleEsp32_SSL_h
#include <BlynkSimpleEsp32_SSL.h>
#endif

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "YourAuthToken";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "YourNetworkName";
char pass[] = "YourPassword";

void setup()
{
  // Debug console
  Serial.begin(9600);

  Blynk.begin(auth, ssid, pass);
}

void loop()
{
  Blynk.run();
}
```

ivankravets, 2018-06-12:
Could you run with `pio run -v`?

ivankravets, 2018-06-15:
- multiple definition of `loop()'
- multiple definition of `setup()'

closed 2018-06-15 by ivankravets